smartdevicelink/sdl_core #1223: HTTP: SDL sends OnSystemRequest "requestType":"LOCK_SCREEN_ICON_URL" on app registered

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [032_ATF_P_TC_PTU_Restarting_Policy_Table_Exchange_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/032_ATF_P_TC_PTU_Restarting_Policy_Table_Exchange_HTTP.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-03-17:
This behavior is correct as far as I can tell, even if it wasn't originally base on a requirement. There would likely need to be several changes to the ATF test scripts to match these expectations, however.

jacobkeeler, 2017-03-21:
@LuxoftAKutsan

LitvinenkoIra, 2017-12-01:
@JackLivio @jacobkeeler so is this behavior is expected? There should be changes in SDL code or in scripts? and it should be mentioned in requirements?

LitvinenkoIra, 2017-12-01:
Priority is Medium.
Not critical redundant SDL activity.

jacobkeeler, 2018-03-08:
Test scripts should be updated, this behavior is expected and was added in response to https://github.com/smartdevicelink/sdl_core/issues/3

closed 2018-03-08 by jacobkeeler