smartdevicelink/sdl_core #1224: ATF: Wrong validation of EXPECT_NOTIFICATION

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [ATF_P_TC_SDL_Build_EXTENDED_POLICY_HTTP](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/ATF_P_TC_SDL_Build_EXTENDED_POLICY_HTTP.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-02-22:
In the [MOBILE_API.xml](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/MOBILE_API.xml#L5125)

```
<param name="url" type="String" maxlength="1000" minsize="1" maxsize="100" array="true" mandatory="false">
```

The `url` parameter has the attribute `array="true"`, which makes it an array of Strings.

This is, oddly, not the case for [HMI_API.xml](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/HMI_API.xml) where it is just defined as a single String (despite its description of `array of URL(s)`). 

It seems that this inconsistency between the two APIs is at least part of the real issue in this case.

jacobkeeler, 2017-03-15:
Likely fixed by #1327, will verify

jacobkeeler, 2017-03-17:
Looks like it was fixed. There was another test failure in this script relating to #1223, however.

closed 2017-03-17 by jacobkeeler