smartdevicelink/sdl_core #1225: SDL does not clear all user consent records in "user_consent_records" section of the LocalPT

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [ATF_Factory_reset.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/user_consent_of_Policies/ATF_Factory_reset.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-03-22:
I wanted to see what the values of "Location-1" and "Notifications" were at the beginning of the test before the app is activated. 

I added these lines to the beginning of the `Test:Precondition_Activate_app()` test
```  
function Test:Precondition_Activate_app()
  testCasesForPolicyTableSnapshot:extract_pts({self.applications[config.application1.registerAppInterfaceParams.appName]})
  local app_consent_location = testCasesForPolicyTableSnapshot:get_data_from_PTS("device_data."..config.deviceMAC..".user_consent_records."..config.application1.registerAppInterfaceParams.appID..".consent_groups.Location-1")
  local app_consent_notifications = testCasesForPolicyTableSnapshot:get_data_from_PTS("device_data."..config.deviceMAC..".user_consent_records."..config.application1.registerAppInterfaceParams.appID..".consent_groups.Notifications")

```

Both app_consent_location and app_consent_notifications are true before user consent is given later in the test function.

Either user_consent for these fields is always true or there is something broken with this test.

aderiabin, 2017-03-23:
> I wanted to see what the values of "Location-1" and "Notifications" were at the beginning of the test before the app is activated.

These data you can find in "sdl_snapshot.json" file which SDL create on start PTU sequence. (Path to it you can find in smartDeviceLink.ini in parameter SystemFilesPath)

> I added these lines to the beginning of the Test:Precondition_Activate_app() test
> ```
> function Test:Precondition_Activate_app() [...]

If you put these lines of code to the beginning of of the `Test:Precondition_Activate_app()` function you will have errors in console during run of script:
```
file doesn't exist: /tmp/fs/mp/images/ivsu_cache/sdl_snapshot.json
file doesn't exist: /tmp/fs/mp/images/ivsu_cache/sdl_snapshot.json
 [31m Element device_data.12ca17b49af2289436f303e0166030a21e525d266e209267433801a8fd4071a0.user_consent_records.0000001.consent_groups.Location-1 is not found in PTS! [0m
 [31m Value of device_data.12ca17b49af2289436f303e0166030a21e525d266e209267433801a8fd4071a0.user_consent_records.0000001.consent_groups.Location-1 is nil [0m
file doesn't exist: /tmp/fs/mp/images/ivsu_cache/sdl_snapshot.json
 [31m Element device_data.12ca17b49af2289436f303e0166030a21e525d266e209267433801a8fd4071a0.user_consent_records.0000001.consent_groups.Notifications is not found in PTS! [0m
 [31m Value of device_data.12ca17b49af2289436f303e0166030a21e525d266e209267433801a8fd4071a0.user_consent_records.0000001.consent_groups.Notifications is nil [0m
```
File "sdl_snapshot.json" has not be created at that moment and variables app_consent_location and app_consent_notifications get default value 0.
 Lua interpret 0 as true.

You can put this code after `Test:Precondition_Activate_app()` function when file "sdl_snapshot.json" is exists and represent LocalPT on start of PTU before activation of application.

aderiabin, 2017-03-23:
Please do not use the script for reproduce current defect due to ATF do not properly 
handle SDL shutdown after BasicCommunication.OnExitAllApplications(reason = "FACTORY_DEFAULTS") notification.
Relevant SDL evolution proposal is created (https://github.com/smartdevicelink/sdl_evolution/pull/73)

JackLivio, 2017-03-23:
@aderiabin Are you saying that the problem is with SDL Core, ATF, or both?

JackLivio, 2017-03-23:
@aderiabin Also when I run this script after a creating a fresh build of sdl core, I do not get those errors you described. I get
```
App Consent location
true
app_consent_notifications
true
```

Note I added some print lines. Here is my activate app test function

```
function Test:Precondition_Activate_app()
  testCasesForPolicyTableSnapshot:extract_pts({self.applications[config.application1.registerAppInterfaceParams.appName]})
  local app_consent_location = testCasesForPolicyTableSnapshot:get_data_from_PTS("device_data."..config.deviceMAC..".user_consent_records."..config.application1.registerAppInterfaceParams.appID..".consent_groups.Location-1")
  local app_consent_notifications = testCasesForPolicyTableSnapshot:get_data_from_PTS("device_data."..config.deviceMAC..".user_consent_records."..config.application1.registerAppInterfaceParams.appID..".consent_groups.Notifications")

  print("App Consent location")
  print(app_consent_location)
  print("app_consent_notifications")
  print(app_consent_notifications)

  local RequestId = self.hmiConnection:SendRequest("SDL.ActivateApp", { appID = self.applications[config.application1.registerAppInterfaceParams.appName]})
  EXPECT_HMIRESPONSE(RequestId)
  :Do(function(_,_)
  local RequestId1 = self.hmiConnection:SendRequest("SDL.GetUserFriendlyMessage", {language = "EN-US", messageCodes = {"DataConsent"}})
  EXPECT_HMIRESPONSE( RequestId1, {result = {code = 0, method = "SDL.GetUserFriendlyMessage"}})
  :Do(function(_,_)
  self.hmiConnection:SendNotification("SDL.OnAllowSDLFunctionality",
  {allowed = true, source = "GUI", device = {id = config.deviceMAC, name = "127.0.0.1", isSDLAllowed = true}})

  local request_id_list_of_permissions = self.hmiConnection:SendRequest("SDL.GetListOfPermissions", { appID = self.applications[config.application1.registerAppInterfaceParams.appName] })
  EXPECT_HMIRESPONSE(request_id_list_of_permissions)
  :Do(function(_,data)
  local groups = {}
  if #data.result.allowedFunctions > 0 then
    for i = 1, #data.result.allowedFunctions do
      groups[i] = data.result.allowedFunctions[i]
      groups[i].allowed = true
    end
  end
commonFunctions:printTable(groups)
  self.hmiConnection:SendNotification("SDL.OnAppPermissionConsent", { appID = self.applications[config.application1.registerAppInterfaceParams.appName], consentedFunctions = groups, source = "GUI"})
  EXPECT_NOTIFICATION("OnPermissionsChange")
  end)
  end)
  end)
```

aderiabin, 2017-03-23:
> @aderiabin  Are you saying that the problem is with SDL Core, ATF, or both?
1. SDL Core has issue described by @KhrystynaDubovyk  in main section. Defect may be reproduced manually passing by preconditions and steps for reproduce.
2. ATF has issue that I described in my previous comment. That is the reason why test script can not be used for reproduce the defect.
> ATF do not properly handle SDL shutdown after BasicCommunication.OnExitAllApplications(reason = "FACTORY_DEFAULTS") notification.

>@aderiabin Also when I run this script after a creating a fresh build of sdl core, I do not get those errors you described.
3. I had run test with your variant of function `Test:Precondition_Activate_app()` with prints and had the same result with errors as I described in previous comment. 
Please check path to sdl_snapshot.json and if it exists remove snapshot and re-run script. This file is created by SDL each time when current scrip has be run.
(Path to it you can find in smartDeviceLink.ini in parameter SystemFilesPath. 
Default it is: /tmp/fs/mp/images/ivsu_cache)

JackLivio, 2017-03-23:
@KhrystynaDubovyk Since I can't use the test script to reproduce this defect I need some clarification on how you came about this issue.

By "LocalPT" do you mean the [policy table contained within CacheManager.cc? ](https://github.com/smartdevicelink/sdl_core/blob/release/4.3.0/src/components/policy/policy_external/src/cache_manager.cc#L98)

If so, when I send a FACTORY_DEFAULTS from the HMI, I have confirmed that ResetUserConsent() in the cache manager is called which clears the user_consent_records.

[See Code Here](https://github.com/smartdevicelink/sdl_core/blob/release/4.3.0/src/components/policy/policy_external/src/cache_manager.cc#L175-L187)

Let me know if this is not related to the original issue.

JackLivio, 2017-03-23:
I think the issue is that the localPT is not backed up before the program shuts itself down.

Would adding backup() to the end of ResetUserConsent() fix the issue?

```
bool CacheManager::ResetUserConsent() {
  CACHE_MANAGER_CHECK(false);
  sync_primitives::AutoLock lock(cache_lock_);
  policy_table::DeviceData::iterator iter =
      pt_->policy_table.device_data->begin();
  policy_table::DeviceData::iterator iter_end =
      pt_->policy_table.device_data->end();

  for (; iter != iter_end; ++iter) {
    iter->second.user_consent_records->clear();
  }
  Backup();
  return true;
}
```

LuxoftAKutsan, 2017-03-24:
@JackLivio , looks like your assumption may be correct. And solution is applicable.

JackLivio, 2017-03-24:
@LuxoftAKutsan I will open a PR

LitvinenkoIra, 2017-12-01:
Priority Hight:
Significant loss of functionality
All data should be cleared during FACTORY RESET

LitvinenkoIra, 2017-12-07:
Fixed in c424760b8a089f507ee93f749f790ee9e594de88 (develop branch)
Works correctly on current develop (9e95f67)
Script for verifying and testing report in attachments [1225.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1539257/1225.tar.gz)

dboltovskyi, 2017-12-20:
Issue was created due to incorrect ATF script 'ATF_Factory_reset.lua'.
In initial version of this script StopSDL() function was envoked before sending of 'BC.OnExitAllApplications' notification with 'FACTORY_DEFAULTS' reason.
This led to the situation when SDL Core was unable to finish factory reset procedure correctly since it was stopped with SIGINT signal.
The Local Polciy Table was not deleted and therefore contains information in ‘device_consent_group and ‘consent_group’ tables
When SDL Core re-started information in these tables was available still.

Afterwards script was corrected in commit https://github.com/smartdevicelink/sdl_atf_test_scripts/commit/2d4f17e104ec75c9c9bc322c67fc25d19336c5f0#diff-683bfcb2db317f63faa337a1c349f575

And then renamed to '221_ATF_Factory_reset.lua' in commit https://github.com/smartdevicelink/sdl_atf_test_scripts/commit/a088063985fc63dbf8fbaad3720acecc8462e515#diff-148e95a8871fab54ec1dbb88c406a9b8

Mentioned in description '1225_FACTORY_DEFAULTS.lua' script is just reworked version of '221_ATF_Factory_reset.lua'

So currently issue is not reproduced on a head of develop (9e95f67178c575ca5a537f9ef3a69653219b073d) as @LitvinenkoIra mentioned eralier.
And this issue can be closed.

mrapitis, 2017-12-21:
Confirming that issue cannot be reproduced on current develop branch and script runs successfully on current develop.

jacobkeeler, 2018-01-08:
I was able to confirm this fix manually and with the `221_ATF_Factory_reset.lua` ATF script. However, when I run the `1225_FACTORY_DEFAULTS.lua` ATF script, it fails during the PTU step. Even though I would consider this issue fixed, the test failure is something to take note of.

JackLivio, 2018-03-01:
Closing since non reproducible. Jacobs issue is due to ATF

closed 2018-03-01 by JackLivio