smartdevicelink/sdl_core #1226: SDL uses url in GetURLs for not registered application

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script  
[010_ATF_P_TC_HMI_sends_GetURLs_one_app_registered.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/010_ATF_P_TC_HMI_sends_GetURLs_one_app_registered.lua)  
[013_ATF_P_TC_HMI_sends_GetURLs_one_app_registered_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/013_ATF_P_TC_HMI_sends_GetURLs_one_app_registered_HTTP.lua)
[015_ATF_P_TC_Sending_PTS_to_Mobile_Application_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/015_ATF_P_TC_Sending_PTS_to_Mobile_Application_HTTP.lua)  
[017_ATF_P_TC_PoliciesManager_Sets_Status_UPDATING_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/017_ATF_P_TC_PoliciesManager_Sets_Status_UPDATING_HTTP.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-02-22:
Edit: added link for [files/jsons/Policies/Policy_Table_Update/few_endpoints_appId.json](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/files/jsons/Policies/Policy_Table_Update/few_endpoints_appId.json)

jacobkeeler, 2017-03-20:
@KhrystynaDubovyk Only the first script that is referenced in this issue should be used in the "-DEXTENDED_POLICY=EXTERNAL_PROPRIETARY" configuration, the rest are meant to be run in the "HTTP" configuration. The preconditions are misleading here if the issue happens in either configuration

LuxoftAKutsan, 2017-12-01:
@jacobkeeler I see merged PR related to this issue. Shouldn't it be closed?

LuxoftAKutsan, 2017-12-01:
Priority is Medium:
Loss of insignificant functionality: List of Urls transferred to HMI but it includes some extra urls

JackLivio, 2018-03-01:
Fixed by #1475

closed 2018-03-01 by JackLivio