smartdevicelink/sdl_core #1228: SDL is using outdated Mobile_API.xml

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script
[018_ATF_P_TC_Sending_PTS_to_mobile_application.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/018_ATF_P_TC_Sending_PTS_to_mobile_application.lua) 
[022_ATF_P_TC_Got_PTU_from_mobile_application.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/022_ATF_P_TC_Got_PTU_from_mobile_application.lua) 
[024_ATF_P_TC_PTU_validation_rules.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/024_ATF_P_TC_PTU_validation_rules.lua)  
[025_ATF_P_TC_PTU_validation_failure.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/025_ATF_P_TC_PTU_validation_failure.lua)  
[053_P_ATF_Register_NewApp_not_exist_inLocalPT_FinishPTU.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/053_P_ATF_Register_NewApp_not_exist_inLocalPT_FinishPTU.lua) 
[054_P_ATF_Register_NewApp_not_exist_inLocalPT_start_PTU_for_NewApp.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/054_P_ATF_Register_NewApp_not_exist_inLocalPT_start_PTU_for_NewApp.lua)  
[032_ATF_P_TC_PTU_Restarting_Policy_Table_Exchange_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/commit/c8060196d50e8178b10bee291dac9b36be0d1e12#diff-9e7b84bc08913ddda5b4b90cc90967ca) 
## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-02-03:
@KhrystynaDubovyk Can you please elaborate on this issue? 

This issue looks to be causing a lot of failed ATF tests and based on the title of the issue, it sounds like an easy fix...

jacobkeeler, 2017-02-22:
@KhrystynaDubovyk [The change for this](https://github.com/smartdevicelink/sdl_core/commit/e1c4aa74dac53a3f81d7199dd99155b9b19fbe07) is currently in both master and develop. If this change actually fixes these tests, then this issue can be closed.

jacobkeeler, 2017-02-22:
For future reference, the API version should not be directly tied to the version of SDL Core

LuxoftAKutsan, 2017-02-27:
@jacobkeeler currently this defect is in develop branch and release 4.3.0. 
Is is related to OnSystemRequest notification url param.
In current develop MOBILE_API it is :

```
 <param name="url" type="String" maxlength="1000" minsize="1" maxsize="100" array="true" mandatory="false">
      <description>
        Optional URL for HTTP requests.
        If blank, the binary data shall be forwarded to the app.
        If not blank, the binary data shall be forwarded to the url with a provided timeout in seconds.
      </description>
</param>
```
HMI_API :
```
      <param name="url" type="String" maxlength="1000" minlength="1" mandatory="false">
        <description>Optional array of URL(s) for HTTP requests.</description>
</param>
```

Note that in HMI_API url param is **not array**.

jacobkeeler, 2017-02-27:
@LuxoftAKutsan Ah, based on the description given it looked like the only issue was the "version" of the Mobile API, not any of its contents.

I noticed this inconsistency as well, and noted it in #1224. I gather that these two issues describe the same problem in that case, are these two issues redundant?

LuxoftAKutsan, 2017-02-27:
@jacobkeeler the issues is about inconsistency between the Mobile API and HMI API. And probably it is something like a duplicate to #1224 
This is fixed with https://github.com/smartdevicelink/sdl_core/pull/1327

jacobkeeler, 2017-02-28:
Closed with #1225

closed 2017-02-28 by jacobkeeler