smartdevicelink/sdl_core #1229: SDL does not increment count_of_rejections_sync_out_of_memory counter after OUT_OF_MEMORY for RAI

KhrystynaDubovyk, 2017-01-24:
## Preconditions
SDL is built with External Proprietary flag

## Steps to reproduce
*Required steps for testing*
1. Create ramdisk with restricted size: 
mkdir /mnt/ramdisk
mount -t tmpfs -o size=300m tmpfs /mnt/ramdisk
 2. Copy to /mnt/ramdisk "bin" folder with SDL.
 3. Start SDL, HMI.
 4. Register app.
 5. Fill app folder via PutFile fully.
 6. Register new App.

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-05-10:
PR: #1486

LuxoftAKutsan, 2017-12-01:
@jacobkeeler shouldn't this issue be closed?

LuxoftAKutsan, 2017-12-01:
Priority is medium:
request is not processed or returned response is wrong in non-standard use case.
Out of memory is not standard use case.

JackLivio, 2018-03-01:
Fixed by#1486

closed 2018-03-01 by JackLivio