smartdevicelink/sdl_core #1253: 'Packed validation failed' in SDL core logs, and SDL corefail to send some RPC's to HMI

madhuy, 2017-02-02:
in case iOS USB, Some times packet validation failing. I see large values in DataSize and MessageID
**DataSize: 2516448294, MessageId: 2976967552**

`DEBUG [31 Dec 2012 20:46:56,723][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:148 RESULT_CODE protocol_handler::ProtocolPacket::ProtocolHeaderValidator::validate: Validating header - Version: 3, Protection: OFF, FrameType: 2, ServiceType: 151, FrameData: 243, SessionId: 237, DataSize: 2516448294, MessageId: 2976967552
INFO  [31 Dec 2012 20:46:56,723][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol/src/service_type.cc:61 ServiceType protocol_handler::ServiceTypeFromByte: Invalid service type: 151
WARN  [31 Dec 2012 20:46:56,723][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:173 RESULT_CODE protocol_handler::ProtocolPacket::ProtocolHeaderValidator::validate: Invalide service type151
WARN  [31 Dec 2012 20:46:56,723][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/incoming_data_handler.cc:153 RESULT_CODE protocol_handler::IncomingDataHandler::CreateFrame: Packet validation failed
DEBUG [31 Dec 2012 20:46:56,723][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/incoming_data_handler.cc:157 RESULT_CODE protocol_handler::IncomingDataHandler::CreateFrame: Malformed message found 1
DEBUG [31 Dec 2012 20:46:56,723][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/incoming_data_handler.cc:164 RESULT_CODE protocol_handler::IncomingDataHandler::CreateFrame: Moved to the next byte 0xb5a49999
TRACE [31 Dec 2012 20:46:56,723][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:94 ProtocolPacket::ProtocolHeader::deserialize: Enter
WARN  [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:127 ProtocolPacket::ProtocolHeader::deserialize: Unknown version:9
TRACE [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:94 ProtocolPacket::ProtocolHeader::deserialize: Exit
DEBUG [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:148 RESULT_CODE protocol_handler::ProtocolPacket::ProtocolHeaderValidator::validate: Validating header - Version: 9, Protection: OFF, FrameType: 7, ServiceType: 243, FrameData: 237, SessionId: 149, DataSize: 4260636337, MessageId: 0
WARN  [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:165 RESULT_CODE protocol_handler::ProtocolPacket::ProtocolHeaderValidator::validate: Unknown version:9
WARN  [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/incoming_data_handler.cc:153 RESULT_CODE protocol_handler::IncomingDataHandler::CreateFrame: Packet validation failed
DEBUG [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/incoming_data_handler.cc:164 RESULT_CODE protocol_handler::IncomingDataHandler::CreateFrame: Moved to the next byte 0xb5a4999a
TRACE [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:94 ProtocolPacket::ProtocolHeader::deserialize: Enter
WARN  [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:127 ProtocolPacket::ProtocolHeader::deserialize: Unknown version:15
TRACE [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:94 ProtocolPacket::ProtocolHeader::deserialize: Exit
DEBUG [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:148 RESULT_CODE protocol_handler::ProtocolPacket::ProtocolHeaderValidator::validate: Validating header - Version: 15, Protection: OFF, FrameType: 3, ServiceType: 237, FrameData: 149, SessionId: 253, DataSize: 4096176496, MessageId: 0
WARN  [31 Dec 2012 20:46:56,724][0xb5cff3e0][ProtocolHandler] /data/sdl_core/src/components/protocol_handler/src/protocol_packet.cc:165 RESULT_CODE protocol_handler::ProtocolPacket::ProtocolHeaderValidator::validate: Unknown version:15`

Attaching logs.
[SmartDeviceLinkCore.log.zip](https://github.com/smartdevicelink/sdl_core/files/748997/SmartDeviceLinkCore.log.zip)

goelgoldi, 2017-02-08:
@joeygrover  Any update on this?

LuxoftAKutsan, 2017-02-09:
@goelgoldi It would be nice if you will provide SDL version (commit), application that is able to reproduce it, environment and flow.

madhuy, 2017-02-09:
@LuxoftAKutsan 
SDL core version https://github.com/smartdevicelink/sdl_core/commit/117fdc9e99d42c7ad3027f25b45c5b8db1e7e78d

Mobile : iPhone 5/6 with iOS 9.3 and 10.2 Connected with USB
HU : Linux arm based

Repro steps:
we have test app which sends some images. While sending image data getting 'Package Validation' failed.

jacobkeeler, 2017-02-17:
I remember running into this issue occasionally (mainly with Spotify) when testing out the AGL port of SDL on a Raspberry Pi 3, though I didn't see the issue when running it on the Porter board. I gather that it may have something to do with overwhelming the transport when running Core on low-end hardware, but I don't know for certain. 

I'm attempting to reproduce the issue now.

LuxoftAKutsan, 2017-02-24:
@madhuy is this issue reproduces always?

madhuy, 2017-03-01:
@LuxoftAKutsan  Yes, it is reproducible with our Hardware. I see this when app sends Images.

LuxoftAKutsan, 2017-03-01:
@madhuy is it possible that there are some malformed data? Can you grab transport data with wire shark?

madhuy, 2017-05-10:
@LuxoftAKutsan Yes, Some data packets losses in transport which causes malformed data. 
As we discussed in workshop How do handle these kind of error cases ? App sends request but never got response because of transport corruption.

LuxoftAKutsan, 2017-12-01:
Priority hight.
Loss of data, the request is not processed with inbound mandatory parameters.

LuxoftAKutsan, 2017-12-11:
@madhuy is it possible to perform get wire shark grabbing data? It would be really useful for investigating

LuxoftAKutsan, 2017-12-11:
@madhuy also could you please describe how did you connect IOS app via USB. We have no iap_transport_adapter_impl.cc in open source. Did you create own transport adapter?

AStasiuk, 2017-12-22:
Issue is related to  proprietary transport adapter and not related to SDL Core.

jacobkeeler, 2018-03-19:
@madhuy Are you still encountering this issue on the latest release?

closed 2018-04-05 by jacobkeeler