smartdevicelink/sdl_core #1255: HMI_API and MOBILE_API issues must be fixed

TMelnyk, 2017-02-03:
**HMI_API:**
1. Re-name DATA_NOT_AVAILABLE resultCode to VEHICLE_DATA_NOT_AVAILABLE
2. Re-name SYNC_OUT_OF_MEMMORY to SYNC_OUT_OF_MEMORY at SystemError enum

**MOBILE_API:**
1 minsize and maxsize must be removed from "url" param of OnSystemRequest notificaton 
2. Add minlenght = 0 to "data" param of OnKeyboardInput notification

JackLivio, 2017-04-18:
@TMelnyk Is there a milestone related to this proposal? If so, could you please add the appropriate milestone?

TMelnyk, 2017-04-18:
@mghiumiusliu , could you please support?

jacobkeeler, 2018-03-02:
Notes for each of these
HMI_API:

  1. Seems like `VEHICLE_DATA_NOT_AVAILABLE` would need to be added to the HMI instead of the existing value being renamed. Both `VEHICLE_DATA_NOT_AVAILABLE` and `DATA_NOT_AVAILABLE` are present in the Mobile API.
  2. This enum seems like it is tied to the Ford SDL implementation, so I wouldn't consider it a high priority (if it were changed, the `SYNC_` prefix should be removed).

MOBILE_API:

  1. I don't think that a limitation of 1000 characters for this parameter is too restrictive, and it would require a major version change to fix this.
  2. Don't know whether this was the case when this issue was made, but this is already the case in master.

In each case (except for the last), if this change is really desired, a proposal should be submitted.

closed 2018-03-02 by jacobkeeler