smartdevicelink/sdl_core #1259: [INI file changes] The default value of "VideoDataStoppedTimeout" and "AudioStoppedTimeout" must be increased to 3000 ms

TMelnyk, 2017-02-06:
**Description:**
1. The default value of "_VideoDataStoppedTimeout_" param must be **increased to 3000ms** at .ini file
2. The default value of "_AudioDataStoppedTimeout_" param must be **increased to 3000ms** at .ini file
3. The new "_iAPBufferSize_" param with 1048576 value **must be added** to [iAP] section of .ini file

**.ini file change:**
_[MEDIA MANAGER]
...
; Defines time in milliseconds for SDL to wait for the next package of raw data over audio service
AudioDataStoppedTimeout = 3000
; Defines time in milliseconds for SDL to wait for the next package of raw data over video service
VideoDataStoppedTimeout = 3000

[IAP]
: Defines size of iAP buffer in bytes
iAPBufferSize = 1048576_ 

**Impact**: 
SDL 
TestCases 
Requirements

JackLivio, 2017-04-18:
@TMelnyk Is this just a proposal or is there a milestone related to this? If so could you please add the appropriate milestone? Thank you

TMelnyk, 2017-04-18:
@JackLivio , this can be closed actually
.ini file is configurable, so any OEM can change valud of any param

closed 2017-04-18 by TMelnyk

JackLivio, 2017-04-19:
@TMelnyk Thanks!