smartdevicelink/sdl_core #1277: [SDL 0014] Adding Audio File Playback to TTSChunk

theresalech, 2017-02-09:
Proposal: [Adding Audio File Playback to TTSChunk](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0014-adding-audio-file-playback-to-ttschunk.md)

> More and more applications use sounds as a means to make a user feel more at home within their application, and allow the developer to have more customization of their User Experience. This proposal will allow developers to provide audio files uploaded to be played in conjunction with, or without, Text-to-Speech text, so that they may provide a more customized experience.

Review: smartdevicelink/sdl_evolution#43

Steering Committee Decision:

> This proposal was accepted. The committee feels it is a good feature to add to give apps some personality and to enable new features, and while the naming isn't ideal, it gives us the maximum amount of flexibility with the minimum extra development cost.  The implementation will have to be fleshed out on core to ensure backward compatibility.

shoamano83, 2017-06-28:
Hi @jacobkeeler,
I think the branch https://github.com/smartdevicelink/sdl_core/tree/feature/audio_file_playback_tts_chunks is for this improvement. I guess this will not be included in SDL Core 4.3, but can I expect that it is included in 4.4 release?

theresalech, 2017-06-29:
Hi @shoamano83, it's possible that this can be included in Core 4.4, but it isn't currently prioritized for the release.  SDLC Members have been tasked with providing prioritized items for the Core 4.4 release, so can include this proposal/issue if needed.

shoamano83, 2017-06-30:
Thanks @theresalech, I understand the situation.

jacobkeeler, 2018-06-18:
Closing with the merge of #1360

PR #1360 (2018-06-14): Add audio file playback capabilities using TTS Chunks
SDL Core implementation of [SDL-0014](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0014-adding-audio-file-playback-to-ttschunk.md)

Resolves #1277 

Documentation PR: smartdevicelink/sdl_hmi_integration_guidelines#58
ATF test updates: smartdevicelink/sdl_atf_test_scripts#1257
closed 2018-06-18 by jacobkeeler