smartdevicelink/sdl_core #1300: SDL returns WARNINGS resultCode in RegisterAppInterface for same - named RC App from different rank devices

aderiabin, 2017-02-16:
## Occurrence:
Always

## Preconditions
SDL is built with RC flag

## Steps to reproduce
1. Start SDL with RC functionality and start HMI
2. Connect 2 mobile devices 
3. Change device rank to DRIVER for device 2
4. Start mobile session on both devices
5. Register Mobile App with next parameters on session on device 1:
appID = "1"
appName ="App1"
6. Register Mobile App with the same parameters (appID and appName) on session on device 2:
appID = "1"
appName ="App1"

## Actual result
SDL returns WARNINGS resultCode in RegisterAppInterface for second registered application (that is, respond with RegisterAppInterface (resultCode: WARNINGS, success: true, params)

## Expected result
SDL must assign the second app with different internal integer appID than the first app has and allow this second app registration (that is, respond with RegisterAppInterface (resultCode: SUCCESS, success: true, params)

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/sdl_rc_functionality (head of branch)
ATF version: 2.2, commit b91f4b89f0230c48c76da082432f61f6bac4a0d5

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-04-18:
@aderiabin Adding RC milestone

LitvinenkoIra, 2017-12-01:
DRIVER rank is not applicable for RC functionality .

closed 2017-12-01 by LitvinenkoIra