smartdevicelink/sdl_core #1301: SDL sends multiple OnHMIStatus notifications instead one required

aderiabin, 2017-02-16:
## Occurrence:
Always

## Preconditions
SDL is built with RC flag

## Steps to reproduce
1. Start SDL with RC functionality and start HMI
2. Connect mobile device 
3. Start mobile session 1
4. Register Mobile App 1 on mobile session 1
5. Start mobile session 2
6. Change device rank to DRIVER for device
7. Activate Mobile App 1
8. Register Mobile App2 on mobile session 2 
9. Activate Mobile App 2

## Actual result
SDL sends multiple OnHMIStatus notifications

## Expected result
SDL sends one OnHMIStatus notifications

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/sdl_rc_functionality (head of branch)
ATF version: 2.2, commit b91f4b89f0230c48c76da082432f61f6bac4a0d5

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-04-18:
@aderiabin Adding RC milestone

LitvinenkoIra, 2017-12-01:
Loss of insignificant functionality: HMI status is received by mobile, any data missed.

HSavynetska, 2018-06-14:
DRIVER rank is not applicable for RC functionality  on current develop(aaadc90)

closed 2018-06-15 by HSavynetska