smartdevicelink/sdl_core #1302: Installing Mac OS X Sierra issue.

rememberkarl, 2017-02-17:
Hi all 'sdl_core' people!

I'm trying to install sdl_core, but I had some issues on it.
First issue was occurred when I tried 'cmake' command. 
'cmake' command was failed because of line number 211 f 'CMakeLists.txt', it was about 'Udev' library.
I just skipped this line and continued installing it, and I had narrowly done it.

But second issue was popped up when I tried 'make' command.
I don't even know what this problem is and requires what kind of solution or library.

Please give me the answers or precise solutions..

Best regards.
Karl

jacobkeeler, 2017-02-20:
Hello @rememberkarl, the reason that you are getting these errors is because the SDL Core project hasn't been tested on Mac OS X, as the project does not officially support building on this system. As it stands, the only supported system for building SDL Core is Ubuntu (though there are plans for a Windows port).

To my knowledge there are no current plans to create a port for OS X in the future, so if you really need to get Core running on your system, you will need to either:
1. Create an Ubuntu Virtual Machine and run SDL Core through that setup, or
2. Write your own OS X port for SDL Core (in which case, it would be appreciated if you commit your port back to this repository upon its completion)

rememberkarl, 2017-02-20:
Hi @jacobkeeler,

Thank you for your comment.
Currently I'm trying to figure out from the beginning which part of this library occurs some issues on Mac OS X, so I'll give you some ideas, hopefully, and after all integration and test on Mac OS X I'll commit my own works for a contribution. (But so many files and codes, you know... ^^;)

Best regards,
Karl

davidswi, 2017-03-20:
Hi all, I have an experimental port of SDL Core to OSX. I am planning to clean it up and submit it as a PR.

davidswi, 2017-03-20:
@rememberkarl I took the approach of exporting the CMake scripts to an XCode workspace as the first step. IMHO, the principal advantage of running SDL Core on OSX is for the rich code analysis and debugging that XCode provides. I also disabled all transports except TCP/IP. I want to clean this up and abstract the BT and USB transport implementations using a PAL that will allow SDL Core to be easily ported to and maintained on OSX, Windows and other platforms that deviate from the Linux device models for USB and BT.

jacobkeeler, 2018-03-02:
Closing this for now, if a MacOS port is desired, a proposal can be submitted to https://github.com/smartdevicelink/sdl_evolution

closed 2018-03-02 by jacobkeeler