smartdevicelink/sdl_core #1305: Unable to reach LIMITED HMI status

vvvakulenko, 2017-02-17:
## Occurrence:
Always
## Preconditions:
1. SDL is started, 
2. SPT is srarted, 
3. phone is connected via USB
## Steps:
1. Add new app
2. Activate app
3. choose info on HMI
## Expected result:
1. app is registered successfully in NONE
2. App is activated successfully in FULL
3. App remains in FULL
## Actual result:
1. app is registered successfully in NONE
2. App is activated successfully in FULL
3. HMI level should be changed to LIMITED
## Environment
OS: Ubuntu 14.04 x64 
SDL repo: https://github.com/smartdevicelink/sdl_core  Head of branch: feature/sdl_rc_functionality
HMI: HTML5 (repo: https://github.com/smartdevicelink/sdl_hmi, HEAD of master branch)
Transport:USB
Mobile device: Huawei U9200
Mobile OS: Android 4.0.3
Mobile App version: 2.1.79, Code:238
## Expected delivery

- [ ] Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-04-18:
@vvvakulenko Do you know if this issue might have to do with the SDL HMI instead of Core?

JackLivio, 2017-04-18:
@vvvakulenko Adding RC milestone

LitvinenkoIra, 2017-12-01:
Significant loss of functionality on SDL side with standard use-case:
HMI level managing is major feature of SDL. And in case if application was deactivated, it should be changed.

LitvinenkoIra, 2017-12-05:
Verified manually
Works correctly on current develop (8c94933)
Logs in attachments 
[SmartDeviceLinkCore.log.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1530726/SmartDeviceLinkCore.log.tar.gz)

LitvinenkoIra, 2017-12-05:
@JackLivio please close this issue

closed 2018-03-01 by JackLivio