smartdevicelink/sdl_core #1306: ResetGlobalProperties does't reset HELPPROMPT and VRHELPITEMS to default values

AStasiuk, 2017-02-17:
Occurrence:

Always

Preconditions:

Delete app_info.dat file.
SDL started. Application is registered and activated on HMI.

Steps to reproduce:

1. Send SetGlobalProperties from mobile (helpPrompt, timeoutPrompt, vrHelpTitle, vrHelp). ==>successfully set
2. Stop smartDeviceLinkCore in terminal (through ctrl+c) and do not exit from mobile app.
3. Start smartDeviceLinkCore and HMI.
4. Mobile App is registered and activated (HMI lvl FULL)
5. Send SetGlobalProperties from mobile (helpPrompt, timeoutPrompt, vrHelpTitle, vrHelp). ==>successfully set
6. Send ResetGlobalProperties from mobile (helpPrompt, timeoutPrompt, vrHelpTitle, vrHelp)

Expected result:

SDL should send to HMI UI.SetGlobalProperties("vrHelpTitle", "vrHelp") with default values.
SDL should send to HMI TTS.SetGlobalProperties with "helpPrompt" parameter that should be reset to default value (smartDeviceLink.ini file contains default values for "helpPrompt" 

Actual result:

SDL sends to HMI UI.SetGlobalProperties(vrHelpTitle) without vrHelp - not OK
SDL sends to HMI TTS.SetGlobalProperties("helpPrompt":[ ]) - empty array - not OK.

Environment

branch: release candidate SDL_RB_B3.10 (87259f84ab14fa2508605c7b8a36730fdd3b910b)
App: any
SPT cw 41 ver:2.1.60
Transport: any
OS: XUbuntu 12.04
HMI: HTML5 

Expected delivery

 Source code updates
 Code comments
 UTs add/update
 ATF tests add/update
 Add/update CI plans/jobs
 SDD updates
 Guidelines update (sdl_core_guides)
 Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-04-18:
@AStasiuk Can you add the related milestone for this issue? I am not sure if it is policies related or remote control. The branch you have listed is an app link branch

Itileda, 2017-04-28:
@JackLivio This SDL issue was reported in scope of smoke testing on release candidate 4.4.0 with REMOTE_CONTROL functionality. Issue is not related to RC.

LitvinenkoIra, 2017-12-01:
Significant loss of functionality on HMI side with standard use-case: 
During ResetGlobalProperties all parameters (vrHelpTitle, helpPromt)  should be processed by SDL properly.

HSavynetska, 2018-06-18:
Defect is not reproduce on current develop(1527db9)
Verified with script(Script under review ): https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1931

Defect is not actual.

HSavynetska, 2018-07-03:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1931
SDL build: develop branch (ba1da7bc2ff3fd37638620f6bea526384646ab0e)
ATF: develop branch

Status: defect is reproduced.

ValeriiMalkov, 2018-07-11:
The next pull request provided bug fixing of this issue:
[PR #2337](https://github.com/smartdevicelink/sdl_core/pull/2337)

jacobkeeler, 2018-07-16:
Note that this will potentially affect behavior in the implementation of #2154 in the case where all existing commands have been deleted, `vr_help` and `help_prompt` will need to be reset.

jacobkeeler, 2018-07-16:
I think based on the description in the [Mobile API](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/MOBILE_API.xml#L3203):
> Resetting means, that they have the same value as at start up (default)

Means that either these default values should be defined by the HMI, or they should be sent by Core at some point automatically. This means if these INI values are being used, these defaults should be used upon registration, not just used when an app sends `ResetGlobalProperties`.

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this issue. Is the review of a fix finished? If so, when will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this issue.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-05-18:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3357

PR #3357 (2020-05-18): ResetGlobalProperties doesn't reset HELPPROMPT and VRHELPITEMS to default values
REUPLOAD #2337 (branch deleted)

Fixes #1306

Fix bug `ResetGlobalProperties` does't reset `HELPPROMPT`
and `VRHELPITEMS` to default values

Fix UT's after bugfixing

 - Fix `ResetGlobalPropertiesRequestTest`
 - Add mock function into `MockApplicationManagerSettings`

Github issue 

UPDATES SINCE #2337

- Change `vr_help_title` smart object type to string instead of map
- Rename unit test variable `SmartObject help_prompt_` to `so_help_prompt` as a variable `help_prompt` already exists and having two so similar is [... cut]
closed 2020-05-18 by iCollin