smartdevicelink/sdl_core #1325: Camera Streaming doesn't work.

rememberkarl, 2017-02-27:
[Delete any non-applicable sections, but we may ask for more information.]

### Bug Report
Camera streaming doesn't work.

##### Reproduction Steps
1. Run RPC Builder.
2. Connect to SDL Core.
3. Go to 'Modules' tab and Streaming item.
4. Check camera mode below 'Video Streaming' module.
5. Select 'Start Streaming' text.

##### Expected Behavior
Camera streaming works and I could see a video streaming on the screen of SDL Core.

##### Observed Behavior
Please check out this link [https://github.com/smartdevicelink/sdl_ios/issues/547](url)

##### OS & Version Information
* iOS Version: 10.2
* SDL Version: 4.2
* Testing Against: SDL Core Simulator connected with iOS device in same Wifi network.

##### Test Case, Sample Code, and / or Example App
Please check out this link [https://github.com/smartdevicelink/sdl_ios/issues/547](url)

JackLivio, 2017-04-18:
@rememberkarl That ios issue doesn't exist? Are you still having a problem

moguli, 2017-04-20:
This issue link is here
https://github.com/smartdevicelink/sdl_ios/issues/547

jacobkeeler, 2018-03-02:
The [updated video streaming guide](https://smartdevicelink.com/en/guides/core/video-streaming-setup/) should be more helpful on this matter, try that and see if you are still having issues.

jacobkeeler, 2018-03-19:
@rememberkarl Closing for now, reopen this if you are still experiencing this issue.

closed 2018-03-19 by jacobkeeler