smartdevicelink/sdl_core #1329: CMake process failing on develop branch with non x86 or x64 archs

mrapitis, 2017-02-27:
The current master branch compiles fine with Raspberry PI architecture, however the current develop fails during the cmake process.  Please command line log below, in addition please see the platform.cmake file that specifically checks for x86 or x64 archs.  The develop branch needs to be updated so that it does not specifically require x86 or x64 archs as this functionality was previously working with master.

https://github.com/smartdevicelink/sdl_core/blob/develop/tools/cmake/helpers/platform.cmake

pi@raspberrypi3:~/core/develop-bin $ cmake ../mycode/
INFO Use gold linker
CMake Warning at CMakeLists.txt:225 (message):
   CCache was not found.
CMake Warning at CMakeLists.txt:238 (message):
   distcc was not found. Distributed compilation is impossible.
-- libudev stable: 1
-- Found UDev: /usr/lib/arm-linux-gnueabihf/libudev.so
--  include: /usr/include
-- USB support enabled
-- Bluetooth support enabled
-- Backtrace support enabled
-- Logger enabled
-- Telemetry monitor enabled
-- PROPRIETARY policy enabled
-- Security enabled
-- HMI PTU decription enabled
CMake Error at tools/cmake/helpers/platform.cmake:87 (message):
  Unsupported architecture
Call Stack (most recent call first):
  tools/cmake/helpers/sources.cmake:85 (get_arch)
  tools/cmake/helpers/sources.cmake:106 (filter_files_by_arch)
  tools/cmake/helpers/sources.cmake:119 (filter_files)
  src/3rd_party-static/jsoncpp/CMakeLists.txt:40 (collect_sources)
-- Configuring incomplete, errors occurred!
See also "/home/pi/core/mycode/CMakeFiles/CMakeOutput.log".
See also "/home/pi/core/mycode/CMakeFiles/CMakeError.log".

LuxoftAKutsan, 2017-03-02:
@mrapitis  Hello, fix available on #1342, but I have not Raspberry pi. Please check it.

JackLivio, 2017-04-18:
@mrapitis closing this since it appears to have been resolved

closed 2017-04-18 by JackLivio