smartdevicelink/sdl_core #1364: UTF-8: Core incorrect handles symbols of two or more bytes size

AStasiuk, 2017-03-10:
Occurrence##

AStasiuk, 2017-03-10:
[SmartDeviceLinkCore.txt](https://github.com/smartdevicelink/sdl_core/files/834428/SmartDeviceLinkCore.txt)

vvvakulenko, 2017-03-20:
**Reproducible**
SDL: HEAD of https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
(commit d5f63b4f66d9ab945d452dfb1c07bf7ee3ca781e)
ATF: HEAD of https://github.com/smartdevicelink/sdl_atf/tree/develop
(commit aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f)

LitvinenkoIra, 2017-12-01:
Priority is medium because Putfile functionality does not work in some specific cases like name of the file contains symbols of two or more bytes and length of file name is more than 255 symbols

GetmanetsIrina, 2018-06-27:
Defect is reproduced on current develop branch(621080dafb543db973d15ba7415c1eb60833a8b0), but with resultCode "GENERIC_ERROR".
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1950.
ATF: develop.

closed 2018-07-10 by ValeriiMalkov