smartdevicelink/sdl_core #1365: [Protocol] SDL respond ACK with protocol version 4 for video and audio services if send start service with 2 or 3 protocol version

AStasiuk, 2017-03-10:
## Occurrence: 
Sometimes (1/5)
## Precondition: 
1. SDL and HMI are started
2. SPT is registered using v2 or v3 protocol
3. Video and Audio services are allowed to be unsecure
4. Wireshark is configured to listen SDL protocol
## Steps:
1. Start VIdeoService.
2. Accept with pop-up on HMI.
3. Start Audio service.
4. Accept with pop-up on HMI.
5. Start RPC service.
## Actual result: 
SDL respond ACK with protocol version 4.
## Expected result: 
SDL should respond ACK with the same protocol as start service was come - version 2.
## Note:
The same behavior if start secure audio/video services.

JackLivio, 2017-04-18:
Duplicate issue https://github.com/smartdevicelink/sdl_core/issues/1034, leaving this one open

jacobkeeler, 2017-11-01:
Closing with the merge of #1630

PR #1630 (2017-09-08): Fix incorrect behaviour of SDL during respose ACK when not rpc service started
Resend #1413 to develop branch

In the case of start not rpc service by low protocol, SDL was answered ACK with highest supported protocol version. It is fairly for RPC service but nit correct for Audio and VIdeo. 

This PR adds check of type service before sending Ack. In case if this is RPC service SDL should respond with maximum supported version. In the case of not RPC service sdl should respond with the same version of protocol as StartService received. 

Fixes : #1365
closed 2017-11-01 by jacobkeeler