smartdevicelink/sdl_core #1376: SDL do not attach binary data to OnSystemRequest notification during Policy update in HTTP mode

LuxoftAKutsan, 2017-03-15:
## Occurrence:
Always

## Preconditions
SDL is built with EXTENDED_POLICY=HTTP flag

## Steps to reproduce
1. Start SDL
2. Perform Policy Update

## Actual result
SDL send OnSystemRequest to Mobile without binary field 

## Expected result
SDL send OnSystemRequest to Mobile with policy table, attached as binary data

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: release/4.3.0 (head of branch)
ATF version: 2.2, commit b91f4b89f0230c48c76da082432f61f6bac4a0d5

## Expected delivery

- [ x]  Source code updates
- [ x]  Code comments
- [ ]x  UTs add/update
- [ x]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

LuxoftAKutsan, 2017-12-01:
Priority high
Significant part of policy update functionality missed

LitvinenkoIra, 2017-12-07:
Works correctly on current develop (9e95f67)
Script for verifying and testing report in attachments [1376.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1538887/1376.tar.gz)

dboltovskyi, 2017-12-20:
Issue is reproduced on commit 458fa67468b2c10a7f8eb18277a83eb90c2f8004 with 'EXTERNAL_PROPRIETARY' SDL flow.
Description changed accordingly.
Test script [1376_PTU_all_flows.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1376_PTU_all_flows.lua) is updated

mrapitis, 2017-12-21:
Current status: issue needs to be retest with updated test script using EXTERNAL_PROPRIETARY flow.

JackLivio, 2018-02-12:
Manually performing policy table update with EXTERNAL_PROPRIETARY works on release 4.5.0. Binary data is attached. Closing issue.

closed 2018-02-12 by JackLivio