smartdevicelink/sdl_core #1377: SDL core ignores the HMI UI.GetCapabilities response  and always use default hmi_capabilities from json file

yang1070, 2017-03-15:
Please see attached log 
[smartdevicelinkcore.zip](https://github.com/smartdevicelink/sdl_core/files/845619/smartdevicelinkcore.zip)

jacobkeeler, 2017-03-15:
At least one of the issues that is causing this is the fact that your softButtonCapabilities is not an array. This value was apparently [changed from an object to an array](https://github.com/smartdevicelink/sdl_core/commit/01ca7ece6ee3f96535b0282ccbe6c89908600c12) in 4.1.0.

The easiest fix here would be to change
```json
"softButtonCapabilities": {
    "imageSupported": true,
    "longPressAvailable": true,
    "shortPressAvailable": true,
    "upDownAvailable": true
}
```
to
```json
"softButtonCapabilities": [ 
    {
        "imageSupported": true,
        "longPressAvailable": true,
        "shortPressAvailable": true,
        "upDownAvailable": true
    }
]
```

jacobkeeler, 2017-03-15:
In addition, I noticed that you have two copies of `showConstantTBTNextTurnIcon` in your `imageFields`.

```json
{
  "imageResolution": {
    "resolutionHeight": 64,
    "resolutionWidth": 64
  },
  "imageTypeSupported": [
    "GRAPHIC_BMP",
    "GRAPHIC_JPEG",
    "GRAPHIC_PNG"
  ],
  "name": "showConstantTBTNextTurnIcon"
},
{
  "imageResolution": {
    "resolutionHeight": 64,
    "resolutionWidth": 64
  },
  "imageTypeSupported": [
     "GRAPHIC_BMP",
     "GRAPHIC_JPEG",
     "GRAPHIC_PNG"
  ],
  "name": "showConstantTBTNextTurnIcon"
}
```

yang1070, 2017-03-16:
Thank you Jacob. Changing the above mentioned content in the ui.getcapabilities response resolves the issue.

closed 2017-03-16 by yang1070