smartdevicelink/sdl_core #1384: [SDLOPEN] SDL doesn't check result codes of HMI IsReady response

istoimenova, 2017-03-20:
Description:
SDL doesn't check result codes of HMI IsReady response

Preconditions
SDL and HMI are started.

Steps to reproduce
At InitHMI_OnReady send IsReady response with (<<error_result_code>>, available = true)

Actual result
SDL accepts only available = true and interface is avalable, SDL doesn't check error_result_code

Expected result
SDL should accept interface if is NOT available

Environment
SDL repo: https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
SDL commit: 7bd5326e8a6f5127220bf938a5be3063bdbb4ab4
OS with SDL: Ubuntu 14.04
VM used for OS with SDL (yes/no): no
ATF version: aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f
Transport: WiFi

Links
Requirements:
[UI Interface] SDL behavior in case HMI does not respond to UI.IsReady_request
[RegisterAppInterface] SDL behavior in case HMI does NOT respond to IsReady request
[UI Interface] HMI does NOT respond to IsReady and mobile app sends RPC that must be splitted

Clarification: Does SDL consider case VR.IsReady response with errorCode is the same as case VR.IsReady is not responded

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to functionality: correctness of result code in response from HMI/SDL

ValeriiMalkov, 2018-07-17:
Questions of this issue: 
[Question #2374](https://github.com/smartdevicelink/sdl_core/issues/2374)

jacobkeeler, 2020-08-24:
Closing with the merge of #3433

PR #3433 (2020-08-24): Add check for result codes from HMI IsReady response
Fixes #1384 and #3205 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
This pull request is intended to fix issue with incorrect determination of interface availability, when HMI sends IsReady response. Corresponding to requirements, message with any error result code and "available" = "true" is equal to "available" = "false", when interface is set as unavailable. That's why check of result codes from IsReady response is added to ChangeInterfaceState method, also minor changes are added to correspon [... cut]
closed 2020-08-24 by jacobkeeler