smartdevicelink/sdl_core #1385: SDL doesn't deactivate app when receives BasicCommunication.OnEventChange(PHONE_CALL,true) from HMI

dboltovskyi, 2017-03-20:
**Occurrence:** Always

**Steps:**
1. Send BasicCommunication.OnEventChanged(Phone_Call,true) from HMI when apps are full 

**Expected result:**
When SDL received BasicCommunication.OnEventChanged(Phone_Call,true) from HMI:
- With Media/VOICE Com app: SDL should send OnHMIStatus(BACKGROUND, NOT_AUDIBLE,MAIN) to mobile app
- With the Navi app: SDL should send OnHMIStatus(LIMITED, NOT_AUDIBLE,MAIN) to mobile app

**Actual result:**
- For VOICE Com app: SDL sends OnHMIStatus(FULL)

**Environment:**
ATF: aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f
SDL: 7bd5326e8a6f5127220bf938a5be3063bdbb4ab4
Parameters: EXTENDED_POLICY=PROPRIETARY

dboltovskyi, 2017-03-20:
[2017-03-20.tar.gz](https://github.com/smartdevicelink/sdl_core/files/855196/2017-03-20.tar.gz)

dboltovskyi, 2017-03-20:
Script&Logs: [2017-03-20_1.tar.gz](https://github.com/smartdevicelink/sdl_core/files/855200/2017-03-20_1.tar.gz)

ValeriiMalkov, 2018-07-30:
ATF test scripts will update for this issue.

dboltovskyi, 2020-06-15:
Issue is still applicable for `6.1` release.
Updated ATF script and logs: [2020-06-15_13-36-35_1385.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4782528/2020-06-15_13-36-35_1385.tar.gz)

JackLivio, 2021-04-21:
Fixed with merge of #2582

PR #2582 (2021-04-21): SDL sends wrong OnHMIStatus for voice app in case of a phone call.
Fixes #1385 

This PR is ready for review.

### Risk
This PR makes no API changes.

### Testing Plan
Covered by ATF script

### Summary
Once SDL received OnEventChanged(Phone_Call, true) from HMI, it should send
OnHMIStatusUpdate with new HMILevel. Expected HMILevel for voice
communication application is "LIMITED", but it is "FULL".

PhoneCallHmiState should take into account voice communication app
as an additional check.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49 [... cut]
closed 2021-04-21 by JackLivio