smartdevicelink/sdl_core #1387: [GENIVI] [Protocol][API] Packet validation failed if send PutFile request from app that uses v.2 protocol and app that uses v.3 protocol also registered

maria-mih, 2017-03-20:
**Preconditions:** 
1. "MaximumPayloadSize" = 1488 in smartDeviceLink.ini file
2. SDL and HMI are started
3. app1 is registered over v.2 protocol, app2 is registered over v.3 protocol
4. app1 is activated 

**Steps to reproduce:**
1. Send PutFile from app1 png_20kb.png (attached)

**Actual result:**
Packet validation failed
SDL does't send any response to the app

**Expected result:**
Request processed successfully
SDL respond with success resultcode
Note:
There is no defect if only app1 (over v.2 protocol) registered

**Environment**
Branch commit: https://github.com/smartdevicelink/sdl_core.git 
SDL branch: feature/extended_policy_functionality 
Commit: 7bd5326e8a6f5127220bf
HMI: HTML5 
OS: 14.04.01 
Virtual machine:no 
ATF commit: N/A
Transport: USB 
Mobile device: Acer Iconia B1-710 
Mobile OS: Android
Mobile App version: Luxoft SPT, version 2.1.85 
Mobile App type (media/non-media/navi):  Media

HSavynetska, 2019-11-19:
The defect is not reproduced in the current develop

Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2048
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop (891497b )
ATF: develop branch

iCollin, 2021-03-31:
Closed as not reproducible against the first 7.1 release candidate.
Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin