smartdevicelink/sdl_core #1389: [GENIVI][API] SDL responds to ListFiles without list of existed files in app folder after re-registration

aderiabin, 2017-03-20:
# Description
SDL responds to ListFiles without list of existed files in app folder after re-registration

## Preconditions
Set proper value to smartDeviceLink.ini for property:
AppStorageFolder = storage
Allow PutFile RPC for FULL HMILevel in policy table.

## Steps to reproduce
- Start SDL and HMI
- Connect mobile and create mobile session
- Register mobile App
- Activate mobile App
- Send RPC "PutFile" from mobile App 
- Unregister mobile App
- Register mobile App
- Send RPC "ListFiles" from mobile App 

## Actual result
 SDL responds to ListFiles without list of existed files in app folder after re-registration.

## Expected result
SDL responds to mobile app with list of existed files in app folder after re-registration.

# Environment
SDL commit: 65ade78808a37560c1a55de058201a10296f14d1
OS with SDL: Ubuntu 14.04
VM used for OS with SDL (yes/no): no
ATF version: 2.2
Transport: WiFi

# Attachments

# Expected delivery
- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Manual tests
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

dboltovskyi, 2017-03-20:
Script: [script.tar.gz](https://github.com/smartdevicelink/sdl_core/files/855430/script.tar.gz)

jacobkeeler, 2017-03-20:
In the script you have, the PutFile that is sent includes `persistentFile=false`, which means that the file shouldn't be kept between sessions.

aderiabin, 2017-03-21:
Parameter `persistentFile` set to `true` just before send PutFile request, please see lines 73-78:
```
  local paramsSend = putFileAllParams()
  paramsSend.syncFileName = "PersistedFile"
  paramsSend.persistentFile = true

  --mobile side: sending PutFile request
  local cid = self.mobileSession:SendRPC("PutFile",paramsSend, "files/icon.png")
```

HSavynetska, 2018-07-04:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1959
SDL build: develop branch ( ba1da7b )
ATF: develop branch

Status: defect is not reproduced.

jacobkeeler, 2018-08-29:
Likely fixed by https://github.com/smartdevicelink/sdl_core/pull/1484, closing for now

closed 2018-08-29 by jacobkeeler