smartdevicelink/sdl_core #1390: SDL sends to Navi app LIMITED, AUDIBLE when user switched to embedded navigation

dboltovskyi, 2017-03-20:
**Occurrence:** Always
**Preconditions:**
1. Core, HMI started
2. App is registered and activated (FULL HMI level)

**Steps:**
1. Switch to embedded navigation: BC.OnEventChanged (eventName = "EMBEDDED_NAVI", isActive = true)

**Expected result:**
SDL sends OnHMIStatus (BACKGROUND, NOT_AUDIBLE) to this media app

**Actual result:**
SDL sends OnHMIStatus (LIMITED, NOT_AUDIBLE) to this media app

**Environment:**
ATF: aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f
SDL: 7bd5326e8a6f5127220bf938a5be3063bdbb4ab4

dboltovskyi, 2017-03-20:
Script&Logs: [2017-03-20_1.tar.gz](https://github.com/smartdevicelink/sdl_core/files/855461/2017-03-20_1.tar.gz)

HSavynetska, 2018-07-12:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1969
SDL build: develop branch ( b7da3b5 )
ATF: develop branch

Status: defect is not reproduced.

dboltovskyi, 2020-06-15:
Issue is not applicable for `6.1` release and can be closed.
Updated ATF script and logs: [2020-06-15_15-56-19_1390.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4782637/2020-06-15_15-56-19_1390.tar.gz)

closed 2020-06-15 by dboltovskyi