smartdevicelink/sdl_core #1392: [GENIVI] Wrong processing of SetDisplayLayout response data

vvvakulenko, 2017-03-20:
# Description
Wrong processing of SetDisplayLayout response data

## Preconditions
SDL is started
Media App is activated

## Steps to reproduce
Send SetDispalyLayout

## Actual result
SetDisplayLayout response from HMI contains parameters: imageFields, imageCapabilities, textFields with value name="timeToDestination". SDL does not send imageCapabilitiesdata to mobile side, name value in imageFields is parsed like next after specified in ImageFieldName enum, name of textFields "timeToDestination" is parsed like 27.

## Expected result 
SDL should send imageCapabilities data to mobile side, name value in imageFields and textFields should be parsed correctly.

## Environment
SDL: HEAD of https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
(commit d5f63b4f66d9ab945d452dfb1c07bf7ee3ca781e)
OS with SDL: Ubuntu 14.04
VM used for OS with SDL: NO
ATF: HEAD of https://github.com/smartdevicelink/sdl_atf/tree/develop
(commit aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f)
Transport: WiFi

vvvakulenko, 2017-03-21:
[Uploading Script_for_ATF_2_1_3_SetDisplayLayout_APPLINK_8959.lua.tar.gz…]()

vvvakulenko, 2017-03-21:
[Uploading SetDisplayLayout_SDLLogs_ATF.zip…]()

HSavynetska, 2018-07-16:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1974
SDL build: develop branch ( 45109a0 )
ATF: develop branch

Status: Defect is reproduced.

iCollin, 2019-10-14:
Closed via #3074

closed 2019-10-14 by iCollin