smartdevicelink/sdl_core #1393: File Names Caching in SDL core

madhuy, 2017-03-20:
SDL core Caches the file names and responds to 'ListFiles' Request from [Cache](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/list_files_request.cc#L80) instead of Enumerating through the file system.

Issue is, There is possibility that System(HU) delete files, in  that case 'ListFiles' response  is invalid.

JackLivio, 2017-03-20:
@madhuy Is core deleting the file or is it some other part of the head unit deleting the file?

madhuy, 2017-03-20:
@JackLivio NOT SDL core. other part of Head unit Native.

madhuy, 2017-03-23:
@JackLivio Please let me know if you have any concerns to enumerating  through file system. 
GetFiles from Cache been called twice.
- from Listfiles
- from resumption apps.

other purpose of cache is to send file [removed notification ](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/delete_file_request.cc#L84) .

goelgoldi, 2017-04-04:
@theresalech this is high priority ticket for Xevo.

madhuy, 2017-05-10:
PR is here https://github.com/smartdevicelink/sdl_core/pull/1484

LitvinenkoIra, 2017-12-04:
@JackLivio is this issue shouldn't be closed? I see PR related to this Issue.

LitvinenkoIra, 2017-12-04:
Works correctly on current develop (8c94933)
Script for verifying and testing report [1393.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1527488/1393.tar.gz)

closed 2017-12-04 by JackLivio