smartdevicelink/sdl_core #1395: [SDLOPEN] SDL doesn't apply sequence SUSPEND -> OnSDLAwake -> SUSPEND -> IGN_OFF for saving resumption data.

istoimenova, 2017-03-21:
Description:
SDL doesn't apply sequence SUSPEND -> OnSDLAwake -> SUSPEND -> IGN_OFF for saving resumption data.

Preconditions:
1. SDL and HMI are started.
2. Application is registered and activated, HMI level = FULL.

Steps to reproduce:
1. Add some data according to data resumption requirements, e.g. AddCommand.
2. Send OnExitAllApplications(SUSPEND).

Actual result:
1. SDL doesn't save data at OnExitAllApplications(SUSPEND).
2. SDL saves data on app_info.dat and sends OnHashChanged notification.

Expected result:
1. SDL shall store all applications resumption-related data on the file system.
2. Send BC.OnPersistanceComplete to HMI on data persistance complete - OK
3. Continue generating resumption related data (in case getting some from mobile application) without sending OnHashChanged notification to mobile app.

Environment:
SDL repo: https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
SDL commit: 7bd5326
OS with SDL: Ubuntu 14.04
VM used for OS with SDL (yes/no): no
ATF version: aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f
Transport: WiFi

Links:
Problem can be reproduced with script:
https://github.com/CustomSDL/sdl_atf_test_scripts/blob/OS_DRAFT/test_scripts/Resumption/ATF_Resumption_PersistentData.lua
Note: script need to be updated.

Requirements:
[Data Resumption]:OnExitAllApplications(SUSPEND) in terms of resumption
[Data Resumption]:OnExitAllApplications(IGNITION_OFF) in terms of resumption

GetmanetsIrina, 2017-12-15:
Reproduced on develop (9e95f67178c575ca5a537f9ef3a69653219b073d)
Works correctly on https://github.com/AKalinich-Luxoft/sdl_core/tree/fix/fix_data_resumption_flow (1f59584)
Script for verifying is https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1395_Resumption_data_IGN_OFF.lua

mrapitis, 2017-12-21:
Reviewed ATF script and ran successfully. Successfully reproduced issue with provided commit. Reviewed code with and requested any changes to public methods require appropriate deprecation.

JackLivio, 2018-01-16:
fix merged

closed 2018-01-16 by JackLivio