smartdevicelink/sdl_core #1396: [Genivi] [TM] Unable to register iOS App via BT.

aderiabin, 2017-03-21:
# Description
Unable to register iOS App via BT.

## Preconditions
- Core, HMI started. 
- iOS phone is paired with SDL's server

## Steps to reproduce
- Start App, BT is used transport
- Try to register App

## Actual result
SDL doesn't see iPhone device, UpdateDeviceList do not bring iOS device

## Expected result
SDL should register App

# Environment
SDL commit: 65ade78808a37560c1a55de058201a10296f14d1
OS with SDL: Ubuntu 14.04
VM used for OS with SDL (yes/no): no
Transport: BT

# Expected delivery
- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Manual tests
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

robinmk, 2017-04-12:
We do not have the feature (iOS apps over BT) yet.

jacobkeeler, 2018-03-08:
Closing because this relates to a feature not yet in the open.

closed 2018-03-08 by jacobkeeler