smartdevicelink/sdl_core #1399: [GENIVI] [API] SDL returns mobile TRUNCATED_DATA with success: false if it received TRUNCATED_DATA from HMI

aderiabin, 2017-03-21:
# Description
SDL returns mobile CANCEL_ROUTE with success: false if it received TRUNCATED_DATA from HMI

## Preconditions
SDL built with policy build flag: PROPRIETARY

## Steps to reproduce
- Start SDL and HMI
- Connect mobile and create mobile session
- Register mobile App
- Activate mobile App
- Perform PTU sequence
- Send RPC "ReadDID" from mobile App 
- Send response on request "VehicleInfo.ReadDID" from HMI with resultCode = "TRUNCATED_DATA"

## Actual result
 SDL sends TRUNCATED_DATA with success: false to mobile

## Expected result
SDL sends TRUNCATED_DATA with success: true to mobile

# Environment
SDL commit: 65ade78808a37560c1a55de058201a10296f14d1
OS with SDL: Ubuntu 14.04
VM used for OS with SDL (yes/no): no
ATF version: 2.2
Transport: WiFi

# Attachments
Script: 

# Expected delivery
- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Manual tests
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

robinmk, 2017-04-12:
The description makes a mention of CANCEL_ROUTE while there is no further mention of it in the steps or results.  Please ensure issue is accurate

JackLivio, 2017-04-18:
@aderiabin  If there is a milestone related to this issue could you please add it?

LitvinenkoIra, 2019-12-12:
Defect is reproduced.
SDL branch: develop(e137556)
ATF: develop
Script: smartdevicelink/sdl_atf_test_scripts#2137

LitvinenkoIra, 2019-12-12:
#2370 - rebased to the latest develop
Fix is valid

LitvinenkoIra, 2019-12-12:
Full regression successfully passed
SDL: fix/wrong_success_code_for_re 38f63bde
Scripts: develop 6f43bfd
ATF: develop 37da7aba

LitvinenkoIra, 2019-12-12:
@theresalech Please triage this issue

theresalech, 2019-12-12:
@LitvinenkoIra We are currently working on finalizing the roadmap with the Steering Committee, so will keep this issue in consideration for upcoming Core releases.

iCollin, 2020-01-14:
Closed via #2370

PR #2370 (2020-01-14): Fix wrong success code for ReadDID request truncated data
Fixes #1399 

This PR is **ready** ready for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Provide ATF tests

### Summary
Fixed SDL sends TRUNCATED_DATA with success: true to mobile

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-14 by iCollin