smartdevicelink/sdl_core #1401: SDL responds with wrong result in case receives GetURLs request from HMI with unknown "service"

dboltovskyi, 2017-03-21:
**Occurrence:** Always
**Preconditions:**
1. Core and HMI started
2. "Endpoint" table in Policy DB has next records for 0x07 service:
7 | http://policies.telematics.ford.com/api/policies | default
3. App is registered on HMI

**Steps:**
1. On HMI click Settings button -> Request GetUrls
2. Enter service id "25"
3. Send GetUrls request -> GetURLs(service 25) is sent to SDL

**Expected result:**
In case SDL receives GetURLs (<service>) request from HMI AND <service> is not defined in "endpoint" table, SDL must repond with SUCCESS result code and without "urls" param

**Actual result:**
SDL responses with "DATA_NOT_AVAILABLE" result ("code" : 9)

**Environment:**
ATF: aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f
SDL: 7bd5326e8a6f5127220bf938a5be3063bdbb4ab4

dboltovskyi, 2017-03-21:
Script&Logs: [2017-03-21_1.tar.gz](https://github.com/smartdevicelink/sdl_core/files/857462/2017-03-21_1.tar.gz)

HSavynetska, 2019-11-19:
The defect is not actual in the current develop. 
GetPolicyConfigurationData API replaces GetURLS API due to changes made in the proposal: https://github.com/smartdevicelink/sdl_evolution/blob/94956699e2db14492a76976c00181fc7a1045c19/proposals/0173-Read-Generic-Network-Signal-data.md

closed 2019-11-19 by HSavynetska