smartdevicelink/sdl_core #1404: [SDLOPEN] SDL cannot start with .ini file in parent folder

istoimenova, 2017-03-21:
Description:
SDL cannot start with .ini file in parent folder

Preconditions:
1. Update in smartDeviceLink.ini: "SDLVersion = {TESTING}"
2. Place -smartDeviceLink.ini file in ../ directory

Steps to reproduce:
1. In console run command ./smartDeviceLinkCore ../

Actual result:
SDL has started successfully, but set default parameters from .ini file. 
Checked with SDL version in SmartDeviceLinkCore.log:
"SDL version: " empty.

Expected result:
SDL has started successfully, set all parameters from .ini file

Environment:
SDL repo: https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
SDL commit: 7bd5326
OS with SDL: Ubuntu 14.04
VM used for OS with SDL (yes/no): no
ATF version: aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f
Transport: WiFi

Links:
Requirements:
[INI file] SDL must start successfully with default settings in case .ini file is omited

Clarification:
Can you clarify that is expected to be "defined directory" for smartDeviceLink.ini

JackLivio, 2017-04-18:
@istoimenova  If there is a milestone related to this issue could you please add it?

AStasiuk, 2020-02-12:
A defect is not actual since a the name of the .ini file should be provided as well (i.e. ./smartDeviceLinkCore ../smartDeviceLinkCore.ini).
With a fully specified name (relative and absolute path) defect is not reproduced.
But here comes another defect: if the specified file does not exist, segmentation fault error will occur.

Environment:
SDL Core: [develop](https://github.com/smartdevicelink/sdl_core/commit/28235f88a6733106a7e8a0d490dfb16a5f374a33)
OS: Ubuntu 16.04

closed 2020-02-12 by AStasiuk