smartdevicelink/sdl_core #1405: [GENIVI] ResetGlobalProperties does't reset HELPPROMPT and VRHELPITEMS to default values

vvvakulenko, 2017-03-21:
# Description
[GENIVI] ResetGlobalProperties does't reset HELPPROMPT and VRHELPITEMS to default values

## Preconditions
1. Delete app_info.dat file.
2. SDL started. Application is registered and activated on HMI.

## Steps to reproduce
1. Send SetGlobalProperties from mobile (helpPrompt, timeoutPrompt, vrHelpTitle, vrHelp). ==>successfully set
2. Stop smartDeviceLinkCore in terminal (through ctrl+c) and do not exit from mobile app.
3. Start smartDeviceLinkCore and HMI.
4. Mobile App is registered and activated (HMI lvl FULL)
5. Send SetGlobalProperties from mobile (helpPrompt, timeoutPrompt, vrHelpTitle, vrHelp). ==>successfully set
6. Send ResetGlobalProperties from mobile (helpPrompt, timeoutPrompt, vrHelpTitle, vrHelp)

## Actual result
SDL sends to HMI UI.SetGlobalProperties(vrHelpTitle) without vrHelp - not OK
SDL sends to HMI TTS.SetGlobalProperties("helpPrompt":[ ]) - empty array - not OK.

## Expected result 
SDL should send to HMI UI.SetGlobalProperties("vrHelpTitle", "vrHelp") with default values.
SDL should send to HMI TTS.SetGlobalProperties with "helpPrompt" parameter that should be reset to default value (smartDeviceLink.ini file contains default values for "helpPrompt" global property).

## Environment
SDL: HEAD of https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
(commit d5f63b4f66d9ab945d452dfb1c07bf7ee3ca781e)
ATF: HEAD of https://github.com/smartdevicelink/sdl_atf/tree/develop
(commit aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f)

JackLivio, 2017-04-18:
@vvvakulenko  If there is a milestone related to this issue could you please add it?

LuxoftAKutsan, 2017-12-07:
Looks like a duplicates : #999  #1405 #1306

jacobkeeler, 2018-03-08:
Duplicate of #1306

closed 2018-03-08 by jacobkeeler