smartdevicelink/sdl_core #1406: [GENIVI] SDL responses StartServiceACK instead of StartServiceNACK if during request for service error occured

vvvakulenko, 2017-03-21:
# Description
SDL responses StartServiceACK instead of StartServiceNACK if during request for service error occured

## Preconditions
1. SDLCore and HMI are started.
2. Mobile application (SPT) is just registered
3. Audio and video services are ONLY allowed to be secure (in smartDeviceLink.ini : ForceProtectedService = 0x0A, 0x0B).
4. The certificate in "module_config" section of PreloadedPT is absent/invalid.

## Steps to reproduce
1. Register and Activate App
2. Click "Start Secure Service" button of Video Service

## Actual result
SDL sent internal error and StartService ACK (encrypted=false)

## Expected result
SDL sent internal error and StartService NACK

JackLivio, 2017-04-18:
@vvvakulenko  If there is a milestone related to this issue could you please add it?

dboltovskyi, 2017-12-25:
Current issue is duplicate of [#1922](https://github.com/SmartDeviceLink/sdl_core/issues/1922) and can be closed.

closed 2017-12-25 by dboltovskyi