smartdevicelink/sdl_core #1408: [GENIVI] After launching app from json the app with the same name from json of second device is removed from AppList

maria-mih, 2017-03-21:
**Occurrence:**
Always

**Preconditions:**
SDL and HMI started
SDL4 functionality is enabled in .ini file.
Devices are connected to SDL.
query_app_response.json is set as a binary file for SystemRequest(QUERY_APPS) from v4 protocol app from both devices.
v4 apps foregrounded on both devices.
SPT Alice on mobiles is configured for starting according to query_app_response.json on both devices.

**Steps to reproduce:**
1. Register apps of v4 protocol on both devices. ==> 6 apps appears on HMI. From them two SPTAlice and Awesome Music App.
2. Activate SPTAlice on HMI.

**Actual result:**
SDL launch SPTAlice thru v4 protocol app on one of device - OK. But after that SDL sends UpdateAppList to HMI with one SPTAlice, which was activated.

**Expected result:**
SDL should send UpdateAppList with activated SPTAlice from one device and SPTAlise from json of second device.

**Environment** 
Branch repository: https://github.com/smartdevicelink/sdl_core.git 
SDL branch: feature/extended_policy_functionality 
Commit: 7bd5326e8a6f5127220bf
HMI: HTML5 
OS: 14.04.01 
Virtual machine:no 
Transport: USB + Bluetooth
Mobile device: Acer Iconia B1-710 and Nexus 5
Mobile OS: Android
Mobile App version: Luxoft SPT, version 2.1.85 
Mobile App type (media/non-media/navi): Media

[Two_devices_HMI.txt](https://github.com/smartdevicelink/sdl_core/files/858749/Two_devices_HMI.txt)
[Two_devicesSmartDeviceLinkCore.txt](https://github.com/smartdevicelink/sdl_core/files/858748/Two_devicesSmartDeviceLinkCore.txt)

JackLivio, 2017-04-18:
@maria-mih If there is a milestone related to this issue could you please add it?

GetmanetsIrina, 2019-11-28:
Issue is not reproduced on current develop https://github.com/smartdevicelink/sdl_core/commit/e137556512215ded94228e33d21ab4c3e1af3017 branch.
Script https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2296
ATF https://github.com/smartdevicelink/sdl_atf/commit/37da7aba8929e14dbd692befdea8540c5dd1dbbb

closed 2019-11-28 by GetmanetsIrina