smartdevicelink/sdl_core #1409: [VS][AS] SDL doesn't send EndService for Video/Audio Service upon IGNITION_OFF

dboltovskyi, 2017-03-22:
**Occurrence:** Always
**Preconditions:**
1. SDL and HMI are started
2. SPT is registered using v3 protocol and activated on HMI

**Steps:**
1. Start Video Service -> SDL send StartServiceACK to mobile
2. Start Audio service -> SDL send StartServiceACK to mobile
3. Send IGNITION_OFF from HMI: On HTML5 HMI click Exit Application and select Ignition Off from drop down

**Expected result:**
SDL sends End Service (Control Frame 0x04) for: RPC, Audio and Video services

**Actual result:**
SDL sends End Service for RPCservice only

**Environment:**
ATF: aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f
SDL: 7bd5326e8a6f5127220bf938a5be3063bdbb4ab4

dboltovskyi, 2017-03-22:
Logs: [logs_2017-03-22.tar.gz](https://github.com/smartdevicelink/sdl_core/files/860581/logs_2017-03-22.tar.gz)

JackLivio, 2017-04-18:
@dboltovskyi If there is a milestone attached with this issue could you please add it to this issue?

mghiumiusliu, 2017-04-19:
@JackLivio this is an old defect. we will fix it in scope of 4.4.1 (regression testing cycle).

HSavynetska, 2018-07-20:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1981
SDL build: develop branch ( 734fb7d )
ATF: develop branch

Status: Defect is reproduced.

theresalech, 2020-04-07:
Closed via #2425

PR #2425 (2020-04-06): Fix SDL doesn't send EndService for Video/Audio Service upon IGNITION_OFF #1409
Fixes #1409 

This PR is ready for review.

### Risk
This PR makes no API changes.

### Testing Plan
Changes in PR are already covered within existing unit tests.

### Summary
Fixed SDL doesn't send EndService for Video/Audio Service upon IGNITION_OFF

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-04-07 by theresalech