smartdevicelink/sdl_core #1452: TLS handshake never finish, cannot start secure video

yang1070, 2017-04-05:
**Occurrence:**
Always

**Preconditions:**
SDL build with -DEXTENDED_MEDIA_MODE=ON

**Steps to reproduce:**
1. in smartdevicelink.ini set Protocol = TLSv1.2; set ForceProtectedService = 0x0A, 0x0B to force secure video/audio for navigation apps.
2. install Ford CA certificates as trusted CAs.
3. Using Android SPT, register a Navigation non-media app via WiFi connection. set app name as "NavTestApp (Android)" and app is as "2160742002"
4. Launch SPT app in HMI once it is found.
5. Launch gstreamer to receive video.
6. in SPT app, start secured video.
7. SPT app will download a certificate for the given app name and app id. 

**Expected result:**
The video starts. SPT log shows the data is encrypted.

**Actual result:**
Video does not start. 

**Environment**
Ubuntu 16.04 x64
Ford SPT 20170321-Nightly-Android.

jacobkeeler, 2018-03-02:
@yang1070 Are you still encountering this issue on the latest release?

jacobkeeler, 2018-03-19:
@yang1070 Closing, reopen this if you are still experiencing this issue.

closed 2018-03-19 by jacobkeeler