smartdevicelink/sdl_core #1479: [SDL 0040] DTLS encryption

theresalech, 2017-04-12:
Proposal: [DTLS encryption](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0040-DTLS-encryption.md)

> This proposal is to add DTLS (Datagram Transport Layer Security) to SDL core. SDL core will utilize the DTLS communications protocol as a preferred solution when sending encrypted data over protected services (such audio 0x0a and video 0x0b ) including the handshake, encryption and decryption processes.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/126

Steering Committee Decision:

> The Steering Committee has agreed to accept this proposal. They've also agreed that we should target Android and iOS library changes as an additional proposal, as this cannot be tested without the library portion.

theresalech, 2018-06-21:
Closed via #2068

closed 2018-06-21 by theresalech

FStoikov, 2019-02-15:
Jira ticket created for current case: https://adc.luxoft.com/jira/browse/FORDTCN-810

PR #2068 (2018-06-18): Feature/dtls encryption
This PR is **ready** for review.

Fixes #1479 

### Risk
This PR makes **no** API changes.

### Testing Plan
Unit-tests are provided

### Summary
Implemented DTLS support

**NOTE**
This feature should be merged **after** #2105 and **before** security fixes #2217 + #2218 

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)