smartdevicelink/sdl_core #1488: Build failed for QT_HMI

moguli, 2017-04-19:
Build commend
cmake -DHMI=qt -DHMI2=ON ../sdl_core-4.2.3/
make

get error messages:
Generating files:
   /home/suy/SSD/SDL/sdl_core_qmi-test/src/components/interfaces/QT_HMI_API.h;/home/suy/SSD/SDL/sdl_core_qmi-test/src/components/interfaces/QT_HMI_API_schema.h
   /home/suy/SSD/SDL/sdl_core_qmi-test/src/components/interfaces/QT_HMI_API_schema.cc
from:
   QT_HMI_API.xml ...

Generating interface source code with following parameters:
    Source xml      : /home/suy/SSD/SDL/sdl_core-4.2.3/src/components/interfaces/QT_HMI_API.xml
    Namespace       : hmi_apis
    Output directory: /home/suy/SSD/SDL/sdl_core_qmi-test/src/components/interfaces
    Parser type     : jsonrpc

Struct 'Common_DeviceInfo' is declared more than once

make[2]: *** [src/components/interfaces/QT_HMI_API.h] Error 22
make[1]: *** [src/components/interfaces/CMakeFiles/HMI_API.dir/all] Error 2
make: *** [all] Error 2

insert0003, 2017-07-03:
I just met same problem.

insert0003, 2017-07-03:
I modified the "src/components/interfaces/QT_HMI_API.xml" like this, this problem had fixed, but still got some other problem.
diff --git a/src/components/interfaces/QT_HMI_API.xml b/src/components/interfaces/QT_HMI_API.xml
index 85a4d41..36e76c9 100644
--- a/src/components/interfaces/QT_HMI_API.xml
+++ b/src/components/interfaces/QT_HMI_API.xml
@@ -1132,20 +1132,6 @@
       <element name="UNKNOWN"/>
       <element name="UNPAIRED"/>
     </enum>
-     <struct name="DeviceInfo">
-      <param name="name" type="String" mandatory="true">
-        <description>The name of the device connected.</description>
-      </param>
-      <param name="id" type="String" mandatory="true">
-        <description>The ID of the device connected</description>
-      </param>
-      <param name="transportType" type="Common.TransportType" mandatory="false">
-        <description>The transport type the named-app's-device is connected over HU(BlueTooth, USB or  WiFi). It must be provided in OnAppRegistered and in UpdateDeviceList</description>
-      </param>
-      <param name="isSDLAllowed" type="Boolean" mandatory="false">
-        <description>Sent by SDL in UpdateDeviceList. ’true’ – if device is allowed for PolicyTable Exchange; ‘false’ – if device is NOT allowed for PolicyTable Exchange </description>
-      </param>
-    </struct>
     <struct name="UserFriendlyMessage" scope="internal">
     <param name="messageCode" type="String" mandatory="true"/>
       <param name="ttsString" type="String" mandatory="false"/>
@@ -3048,7 +3034,7 @@
       <description>true, if successful</description>
       <description>false, if failed</description>
       </param>
-      <param name="resultCode" type="Result" platform="documentation">
+      <param name="resultCode" type="Common.Result" platform="documentation">
         <description>See Result</description>
       <element name="SUCCESS"/>
       <element name="INVALID_DATA"/>

jacobkeeler, 2017-10-19:
See https://github.com/smartdevicelink/sdl_core/issues/1824 for more details about the QT HMI (Bottom line is that it hasn't been supported in a long time, and it is not yet clear whether we will in the future, I hope to have more details sometime in November).

jacobkeeler, 2018-03-02:
The QT HMI option will be removed in the next release (https://github.com/smartdevicelink/sdl_core/issues/1857), though a custom version of the HMI could be connected using web sockets.

closed 2018-03-02 by jacobkeeler