smartdevicelink/sdl_core #1493: [SDL 0043] Move to the new C++11 standard

theresalech, 2017-04-19:
Proposal: [Move to the new C++11 standard](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0043-upgrade-c++-standard.md)

> This proposal is geared to upgrade currently used `C++98` standard to `C++11` as the latter one has much more useful features and significant improvements from the performance point of view. Also it became more safe in runtime and allows to catch a lot of issues during the compilation stage.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/132

Steering Committee Decision:

> The Steering Committee has agreed to accept this proposal, and noted that we will need to identify implementation timing and clarify LTS before implementation.

jacobkeeler, 2018-08-30:
Potential improvements to project using C++11 features (from proposal review):
- [ ] Move semantics and rvalue reference.
- [ ] Move constructors and assignment operators.
- [ ] Variadic templates
- [ ] Initializer list
- [ ] Static assertion
- [x] Auto typed variables (should be done as new code is added)
- [ ] SFINAE
- [ ] Strongly typed enums
- [ ] Default and deleted function
- [x] smart pointers - std::unique_ptr and std::shared_ptr

theresalech, 2021-09-21:
Closing as implemented in Core 5.0.

closed 2021-09-21 by theresalech