smartdevicelink/sdl_core #1495: Build failing on Fedora 25

zeenix, 2017-04-20:
It would seem that SDL's API usage is incompatible with log4cxx (0.10.0) in Fedora 25:

```
In file included from /home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:40:0,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_schema.h:35,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_object.h:41,
                 from /home/zeenix/checkout/sdl_core/src/components/formatters/include/formatters/CSmartFactory.h:38,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.h:42,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.cc:42:
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:89:65: error: ‘static const LogString log4cxx::spi::LoggingEvent::getCurrentThreadName()’ is private within this context
               ::log4cxx::spi::LoggingEvent::getCurrentThreadName()); \
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:122:3: note: in expansion of macro ‘LOG_WITH_LEVEL’
   LOG_WITH_LEVEL(loggerPtr, ::log4cxx::Level::getFatal(), logEvent)
   ^~~~~~~~~~~~~~
/home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:94:5: note: in expansion of macro ‘LOG4CXX_FATAL’
     LOG4CXX_FATAL(logger_,                                                    \
     ^~~~~~~~~~~~~
/home/zeenix/checkout/sdl_core/src/components/utils/include/utils/convert_utils.h:68:3: note: in expansion of macro ‘DCHECK_OR_RETURN’
   DCHECK_OR_RETURN(value >= std::numeric_limits<OutputType>::min(),
   ^~~~~~~~~~~~~~~~
In file included from /home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:41:0,
                 from /home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:40,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_schema.h:35,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_object.h:41,
                 from /home/zeenix/checkout/sdl_core/src/components/formatters/include/formatters/CSmartFactory.h:38,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.h:42,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.cc:42:
/usr/include/log4cxx/spi/loggingevent.h:247:47: note: declared private here
                        static const LogString getCurrentThreadName();
                                               ^~~~~~~~~~~~~~~~~~~~
In file included from /home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:40:0,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_schema.h:35,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_object.h:41,
                 from /home/zeenix/checkout/sdl_core/src/components/formatters/include/formatters/CSmartFactory.h:38,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.h:42,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.cc:42:
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:89:65: error: ‘static const LogString log4cxx::spi::LoggingEvent::getCurrentThreadName()’ is private within this context
               ::log4cxx::spi::LoggingEvent::getCurrentThreadName()); \
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:122:3: note: in expansion of macro ‘LOG_WITH_LEVEL’
   LOG_WITH_LEVEL(loggerPtr, ::log4cxx::Level::getFatal(), logEvent)
   ^~~~~~~~~~~~~~
/home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:94:5: note: in expansion of macro ‘LOG4CXX_FATAL’
     LOG4CXX_FATAL(logger_,                                                    \
     ^~~~~~~~~~~~~
/home/zeenix/checkout/sdl_core/src/components/utils/include/utils/convert_utils.h:70:3: note: in expansion of macro ‘DCHECK_OR_RETURN’
   DCHECK_OR_RETURN(value <= std::numeric_limits<OutputType>::max(),
   ^~~~~~~~~~~~~~~~
In file included from /home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:41:0,
                 from /home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:40,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_schema.h:35,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_object.h:41,
                 from /home/zeenix/checkout/sdl_core/src/components/formatters/include/formatters/CSmartFactory.h:38,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.h:42,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.cc:42:
/usr/include/log4cxx/spi/loggingevent.h:247:47: note: declared private here
                        static const LogString getCurrentThreadName();
                                               ^~~~~~~~~~~~~~~~~~~~
In file included from /home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:40:0,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_schema.h:35,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_object.h:41,
                 from /home/zeenix/checkout/sdl_core/src/components/formatters/include/formatters/CSmartFactory.h:38,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.h:42,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.cc:42:
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:89:65: error: ‘static const LogString log4cxx::spi::LoggingEvent::getCurrentThreadName()’ is private within this context
               ::log4cxx::spi::LoggingEvent::getCurrentThreadName()); \
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:122:3: note: in expansion of macro ‘LOG_WITH_LEVEL’
   LOG_WITH_LEVEL(loggerPtr, ::log4cxx::Level::getFatal(), logEvent)
[... 113 lines cut ...]
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:122:3: note: in expansion of macro ‘LOG_WITH_LEVEL’
   LOG_WITH_LEVEL(loggerPtr, ::log4cxx::Level::getFatal(), logEvent)
   ^~~~~~~~~~~~~~
/home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:94:5: note: in expansion of macro ‘LOG4CXX_FATAL’
     LOG4CXX_FATAL(logger_,                                                    \
     ^~~~~~~~~~~~~
/home/zeenix/checkout/sdl_core/src/components/utils/include/utils/convert_utils.h:70:3: note: in expansion of macro ‘DCHECK_OR_RETURN’
   DCHECK_OR_RETURN(value <= std::numeric_limits<OutputType>::max(),
   ^~~~~~~~~~~~~~~~
In file included from /home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:41:0,
                 from /home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:40,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_schema.h:35,
                 from /home/zeenix/checkout/sdl_core/src/components/smart_objects/include/smart_objects/smart_object.h:41,
                 from /home/zeenix/checkout/sdl_core/src/components/formatters/include/formatters/CSmartFactory.h:38,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.h:42,
                 from /home/zeenix/checkout/sdl_core/src/components/interfaces/MOBILE_API_schema.cc:42:
/usr/include/log4cxx/spi/loggingevent.h:247:47: note: declared private here
                        static const LogString getCurrentThreadName();
                                               ^~~~~~~~~~~~~~~~~~~~
src/components/interfaces/CMakeFiles/MOBILE_API.dir/build.make:111: recipe for target 'src/components/interfaces/CMakeFiles/MOBILE_API.dir/MOBILE_API_schema.cc.o' failed
make[2]: *** [src/components/interfaces/CMakeFiles/MOBILE_API.dir/MOBILE_API_schema.cc.o] Error 1
CMakeFiles/Makefile2:1054: recipe for target 'src/components/interfaces/CMakeFiles/MOBILE_API.dir/all' failed
make[1]: *** [src/components/interfaces/CMakeFiles/MOBILE_API.dir/all] Error 2

```

gunnarx, 2017-04-20:
I would follow up with the following to hopefully bring us forward quickly.

1. Could we be informed which version of log4cxx is expected by sdl_core?  In case we must compile it separately in order to fill the dependency with something that works.  No matter what, we need to put this on the embedded system shortly after this, so we'll need to fill all dependencies independent of any host system.  Compiling on a Fedora host is only a partial step along the way.

2. What system(s) do SDL developers usually use for their day-to-day development?

gunnarx, 2017-04-20:
By the way @zeenix, was this the tip of master or some other branch/tag/commit?

zeenix, 2017-04-20:
Current git master: e4b15fea129d9061e87e2fce5d0e99dbd4383ed62.

jacobkeeler, 2017-04-20:
@gunnarx To answer your questions
1. sdl_core expects that you don't have log4cxx installed through your package manager at all. The 
log4cxx library is configured/built within the sdl_core project itself and installed during this process. You shouldn't need to build the library separately, some conflicts just seems to occur when log4cxx is installed through the package manager rather than this project (not entirely certain why this is, but it may have to do with configuration differences). This fact is noted in the "Known dependency issues" section of the README.
2. The only officially supported system for running the sdl_core project is Ubuntu 14.04, although it works fine in 16.04 as well in my experience (just a few deprecation warnings which are connected to the newer version of GCC included). Most other Linux systems should be able to build the project, but they haven't been tested and therefore may have minor issues during the build.

zeenix, 2017-04-20:
If log4cxx is not installed on my system, I get the following errors in multiple locations:

```
src/components/transport_manager/CMakeFiles/transport_manager.dir/build.make:86: recipe for target 'src/components/transport_manager/CMakeFiles/transport_manager.dir/src/transport_manager_default.cc.o' failed
make[2]: *** [src/components/transport_manager/CMakeFiles/transport_manager.dir/src/transport_manager_default.cc.o] Error 1
CMakeFiles/Makefile2:1278: recipe for target 'src/components/transport_manager/CMakeFiles/transport_manager.dir/all' failed
make[1]: *** [src/components/transport_manager/CMakeFiles/transport_manager.dir/all] Error 2
[300%] Performing install step for 'intergen'
In file included from /home/zeenix/checkout/sdl_core/src/components/include/utils/macro.h:40:0,
                 from /home/zeenix/checkout/sdl_core/src/components/include/utils/lock.h:42,
                 from /home/zeenix/checkout/sdl_core/src/components/connection_handler/include/connection_handler/connection.h:39,
                 from /home/zeenix/checkout/sdl_core/src/components/connection_handler/src/connection.cc:37:
/home/zeenix/checkout/sdl_core/src/components/include/utils/logger.h:40:42: fatal error: log4cxx/propertyconfigurator.h: No such file or directory
 #include <log4cxx/propertyconfigurator.h>
                                          ^
compilation terminated.

```
Since you said log4cxx doesn't come from system, I suspected it could be a parallel build issue and it seems to be the case since if I run `make -j1`, I don't get that error but instead see another error:

```
home/zeenix/checkout/sdl_core/src/3rd_party/apache-log4cxx-cmake/../apache-log4cxx-0.10.0/src/main/cpp/locationinfo.cpp: In member function 'void log4cxx::spi::LocationInfo::write(log4cxx::helpers::ObjectOutputStream&, log4cxx::helpers::Pool&) const':
/home/zeenix/checkout/sdl_core/src/3rd_party/apache-log4cxx-cmake/../apache-log4cxx-0.10.0/src/main/cpp/locationinfo.cpp:163:21: error: narrowing conversion of '237' from 'int' to 'char' inside { } [-Wnarrowing]
          0x78, 0x70 };
                     ^
/home/zeenix/checkout/sdl_core/src/3rd_party/apache-log4cxx-cmake/../apache-log4cxx-0.10.0/src/main/cpp/locationinfo.cpp:163:21: error: narrowing conversion of '153' from 'int' to 'char' inside { } [-Wnarrowing]
/home/zeenix/checkout/sdl_core/src/3rd_party/apache-log4cxx-cmake/../apache-log4cxx-0.10.0/src/main/cpp/locationinfo.cpp:163:21: error: narrowing conversion of '187' from 'int' to 'char' inside { } [-Wnarrowing]
/home/zeenix/checkout/sdl_core/src/3rd_party/apache-log4cxx-cmake/../apache-log4cxx-0.10.0/src/main/cpp/locationinfo.cpp:163:21: error: narrowing conversion of '225' from 'int' to 'char' inside { } [-Wnarrowing]
/home/zeenix/checkout/sdl_core/src/3rd_party/apache-log4cxx-cmake/../apache-log4cxx-0.10.0/src/main/cpp/locationinfo.cpp:163:21: error: narrowing conversion of '145' from 'int' to 'char' inside { } [-Wnarrowing]
/home/zeenix/checkout/sdl_core/src/3rd_party/apache-log4cxx-cmake/../apache-log4cxx-0.10.0/src/main/cpp/locationinfo.cpp:163:21: error: narrowing conversion of '165' from 'int' to 'char' inside { } [-Wnarrowing]

```
This is about new version of gcc catching problems combined with warnings being treated as errors. Unless you always test your code against latest versions of gcc, I'd strongly recommend you don't treat warnings as errors.

I completely understand that it's impossible to test on all Linux distributions out there but it would be nice if latest releases of most popular few distributions (e.g Ubuntu, Fedora and Debian) are supported.

gunnarx, 2017-04-20:
@jacobkeeler thanks for quick answer.   We're just trying to get up and running quickly, and I just saw the comment in the README also.  While it doesn't "seem right" to have to remove log4cxx from the system, we'll go with the flow here.  Appreciate you taking the time - we'll need a little guidance probably and might get on slack also.

phongt, 2017-04-20:
@zeenix   Please check these 2 patches. I ported from logcxx upstream for fixing building failure of gcc 6.x
[0003-Applied-patch-for-LOGCXX-400-LOGCXX-404-LOGCXX-402-L.patch](https://github.com/phongt/meta-sdl/blob/release/4.1.0/recipes-devtools/log4cxx/log4cxx/0003-Applied-patch-for-LOGCXX-400-LOGCXX-404-LOGCXX-402-L.patch)
[svn-log4cxx-rev-1751050.patch](https://github.com/phongt/meta-sdl/blob/release/4.1.0/recipes-devtools/log4cxx/log4cxx/svn-log4cxx-rev-1751050.patch)

zeenix, 2017-04-20:
@phongt Thanks so much. That helps! At least the build is continuing now.

zeenix, 2017-04-21:
After a successful `make` and `make install` I cant' run it:

$ LD_LIBRARY_PATH=. ./smartDeviceLinkCore 
./smartDeviceLinkCore: error while loading shared libraries: liblog4cxx.so.10: cannot open shared object file: No such file or directory

zeenix, 2017-04-21:
Found where the library was and managed to run the binary but it crashes soon after:

```
$ LD_LIBRARY_PATH=.:../src/3rd_party/apache-log4cxx-build/src/main/cpp/.libs/ ./smartDeviceLinkCore 
log4cxx: setFile(SmartDeviceLinkCore.log,true) call failed.
log4cxx: IO Exception : status code = 13
log4cxx: setFile(ProtocolFordHandling.log,false) call failed.
log4cxx: IO Exception : status code = 13
log4cxx: setFile(TransportManager.log,false) call failed.
log4cxx: IO Exception : status code = 13
log4cxx: No output stream or file set for the appender named [SmartDeviceLinkCoreLogFile].
smartDeviceLinkCore: /home/zeenix/checkout/sdl_core/src/components/include/utils/shared_ptr.h:322: ObjectType* utils::SharedPtr<ObjectType>::operator->() const [with ObjectType = resumption::ResumptionData]: Assertion `(mObject)' failed.
Aborted (core dumped)

```
The backtrace:

```
smartDeviceLinkCore: /home/zeenix/checkout/sdl_core/src/components/include/utils/shared_ptr.h:322: ObjectType* utils::SharedPtr<ObjectType>::operator->() const [with ObjectType = resumption::ResumptionData]: Assertion `(mObject)' failed.

Thread 1 "MainThread" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
58	}
Missing separate debuginfos, use: dnf debuginfo-install nss-softokn-freebl-3.29.3-1.0.fc25.x86_64
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
#1  0x00007ffff554b51a in __GI_abort () at abort.c:89
#2  0x00007ffff5541da7 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x1b96181 "(mObject)", 
    file=file@entry=0x1b96138 "/home/zeenix/checkout/sdl_core/src/components/include/utils/shared_ptr.h", line=line@entry=322, 
    function=function@entry=0x1b96420 <utils::SharedPtr<resumption::ResumptionData>::operator->() const::__PRETTY_FUNCTION__> "ObjectType* utils::SharedPtr<ObjectType>::operator->() const [with ObjectType = resumption::ResumptionData]") at assert.c:92
#3  0x00007ffff5541e52 in __GI___assert_fail (assertion=0x1b96181 "(mObject)", 
    file=0x1b96138 "/home/zeenix/checkout/sdl_core/src/components/include/utils/shared_ptr.h", line=322, 
    function=0x1b96420 <utils::SharedPtr<resumption::ResumptionData>::operator->() const::__PRETTY_FUNCTION__> "ObjectType* utils::SharedPtr<ObjectType>::operator->() const [with ObjectType = resumption::ResumptionData]") at assert.c:101
#4  0x00000000012dd9c3 in utils::SharedPtr<resumption::ResumptionData>::operator-> (this=0x22a3318)
    at /home/zeenix/checkout/sdl_core/src/components/include/utils/shared_ptr.h:322
#5  0x00000000012d2ae8 in resumption::ResumeCtrlImpl::OnSuspend (this=0x22a31c0)
    at /home/zeenix/checkout/sdl_core/src/components/application_manager/src/resumption/resume_ctrl_impl.cc:268
#6  0x000000000120edb2 in application_manager::ApplicationManagerImpl::UnregisterAllApplications (this=0x22a1550)
    at /home/zeenix/checkout/sdl_core/src/components/application_manager/src/application_manager_impl.cc:2515
#7  0x0000000001201807 in application_manager::ApplicationManagerImpl::Stop (this=0x22a1550)
    at /home/zeenix/checkout/sdl_core/src/components/application_manager/src/application_manager_impl.cc:1767
#8  0x00000000011cce1a in main_namespace::LifeCycle::StopComponents (this=0x7fffffffccf0)
    at /home/zeenix/checkout/sdl_core/src/appMain/life_cycle.cc:348
#9  0x00000000011c49bd in main (argc=1, argv=0x7fffffffddf8) at /home/zeenix/checkout/sdl_core/src/appMain/main.cc:153

```

phongt, 2017-04-21:
@zeenix Perhaps 'sudo make install' caused this issue. I would like to share my experience for building and run SDL core (my environment is Ubuntu 16.04.2 LTS).

$ git clone https://github.com/smartdevicelink/sdl_core.git
$ mkdir -p build
$ cd build
$ cmake ../sdl_core/
$ make
$ make install
$ cd bin
$ LD_LIBRARY_PATH=. ./smartDeviceLinkCore
Hope this can help.

jacobkeeler, 2017-04-21:
@zeenix Also, the `log4cxx library.so.10` that it was unable to find should have been copied to your `/usr/local/lib` directory during the install process. You may need to add this directory to your `/etc/ld.so.conf`, then run `ldconfig` in order for the library to be found correctly.

As for your most current issue, the permissions issue should be easily fixed by removing the `bin` folder in your SDL build directory (`rm -r bin/`). Then you should be able to just run `make install` again (WITHOUT `sudo`) and running SDL core should work from there.

zeenix, 2017-04-24:
Thanks @phongt and @jacobkeeler ! the permission issue was more about `make` rather than `make install` needing sudo access. Also how would `make install` install in /usr/local without elevated privileges?

zeenix, 2017-04-24:
With some minor corrections to the steps kindly provided by @phongt, I'm now able to successfully run SDL core. I tried the reference sdl_hmi to test but I failed to find a way to make the HMI actually connect to my phone. Althgouh I do see it connecting to SDL core on the console:

```
$ LD_LIBRARY_PATH=/usr/local/lib:. ./smartDeviceLinkCore
accept_buf: XYKnFI3+m2rW/o9dq1uvZdFby6E=
accept_buf: MGsLaHENiRvkNbUNcb3ncB/iqvA=
accept_buf: xLpshPmfZNSilqZ6XtppYD7qKno=
accept_buf: GYeLGOr5tE0l3PcukgQtUwKKxqI=
accept_buf: jbpi5/HL11sr9DTazRjGorod+6U=
accept_buf: 3guRNxo667xvQ0AFsdAHwUzqyvM=
accept_buf: Uhx+CC8qJf2FAO/tiFNB5IN0yWA=
ERROR!!! /home/zeenix/checkout/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:96 Received 4096 actual data length 4916
```

jacobkeeler, 2017-04-24:
@zeenix to answer your earlier question, `make install` doesn't require sudo access because it really only installs files into your `$SDL_BUILD_DIR/bin` folder. As you pointed out in #1499, the installation of the logger really should be done in this step, but it is currently done in the `make` step. I've been looking into how to fix that issue, but it looks like it may not have a quick fix.

phongt, 2017-04-25:
@zeenix  For your information
> ERROR!!! /home/zeenix/checkout/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:96 Received 4096 actual data length 4916

The issue were raised in #913

jacobkeeler, 2017-04-25:
@zeenix what app are you trying to connect?

zeenix, 2017-04-25:
@jacobkeeler My phone but i probably just need to read some docs first on how to go about that? @gunnarx told me that I've to connect from the phone somehow using some android app perhaps?

jacobkeeler, 2017-04-25:
@zeenix SDL Core will connect to any app that is SDL-connected, not just the phone on its own. There are a few production apps which you could use such as Spotify or Pandora, or you could use one of our test/sample apps (I would suggest trying the RPC builder app for [Android](https://github.com/smartdevicelink/rpc_builder_app_android) or [iOS](https://github.com/smartdevicelink/rpc_builder_app_ios)).

Reshma9916, 2017-06-07:
after ./smartDeviceLink. what is the output. it is using like this
.accept_buf: u0mB4cezIkf558WRHxq3i6z+1uo=
accept_buf: 1M5fd8EAtklTzRI9u61UedTQyKw=
accept_buf: J3NlTyNRrZuSwGUgp8nCJH4xFMk=
accept_buf: IXoPakVjyNEhyCObcG8tIL3az7U=
accept_buf: S+11m8RiFrTQpPUyNAPnU5UiF30=
accept_buf: ea8UwXB2Pk2ZerD2hRNgeRqmaok=
accept_buf: nFo39KiI5mpPaqHswG0i08E6dEY=
accept_buf: maUMf2zZnj55KdDeTQj06s2G9Jw=
ERROR!!! /u/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:97 Incomplete messageERROR!!! /u/sdl_core/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp:96 Received 4096 actual data length 4918
  And in HMI, it is not able to list the apps but it is showing the Connected device(mobile connected through USB).

jacobkeeler, 2018-04-09:
Fixed with the merge of #2097

PR #2097 (2018-04-09): Fix build failures when using GCC7

Fixes #1495 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Fixes `narrowing` warnings in log4cxx, as well as an incorrect return type in the security manager

### Changelog
##### Bug Fixes
* Fix `narrowing` warnings in log4cxx due to conversion from `int` literals to `char`
* Fix implicit conversion of `HandshakeResult` to `bool` in return statement for `WriteHandshakeData`

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-04-09 by jacobkeeler