smartdevicelink/sdl_core #1518: Segment Fault when close Navigation app

yang1070, 2017-05-02:
To reproduce: connect SyncProxyTester as Navigation app, repeat the procedure:
launch SyncProxyTester from HMI
start unsecured video in SPT
stop unsecured videe in SPT
start unsecured video in SPT
stop unsecured videe in SPT
exit SyncProxyTester from HMI menu

I see about 50 timer threads running or scheduled
[image]
 to run void ApplicationManagerImpl::CloseNaviApp()

shoamano83, 2017-05-02:
Hi @yang1070,  I faced similar crash in application_manager::ApplicationManagerImpl::CloseNaviApp() while running a video streaming.

Perhaps we need:
```
  if (!navi_app_to_stop_.empty()) {
    :
    :  
  }
```
condition check before running `uint32_t app_id = navi_app_to_stop_.front();` ?

shoamano83, 2017-05-08:
Hi @yang1070, it looks like the crash issue has been already fixed in develop branch. I built a SDL core binary from develop branch and didn't see my crash anymore.
I think this is the fix: https://github.com/smartdevicelink/sdl_core/commit/0ea564a51da6aaa4ddca5e0b52bf9471dd586bf4

So I think you can try a build using latest develop branch and see if your issues still exist.

jacobkeeler, 2018-03-08:
@yang1070 Closing for now since it appears fixed, feel free to reopen if it resurfaces

closed 2018-03-08 by jacobkeeler