smartdevicelink/sdl_core #1550: [SDL 0048] Add H.264 over RTP format support for video streaming

theresalech, 2017-05-17:
Proposal: [Add H.264 over RTP format support for video streaming](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0048-H264-over-RTP-support-for-video-streaming.md)

> This proposal aims to achieve more robust and smooth video streaming for navigation and projection by adding timestamp information to a video stream.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/143

Steering Committee Decision:

> The Steering Committee has agreed to accept this proposal with revisions based on the outcome of [#176](https://github.com/smartdevicelink/sdl_evolution/issues/176) (how the flow should work, enum versus string format).

Toyota-Sbetts, 2017-08-15:
@theresalech The core changes necessary to support this were made in the video streaming capabilities proposal #1700.  Otherwise, RTP streaming changes will only happen on the Proxy side and HMI side.  If the project maintainer agrees, I think we can close this ticket in the Core repository.

shoamano83, 2017-08-16:
That's true. The initial version of RTP proposal included some of video capability idea, but they were moved to Video Streaming Capabilities proposal. Thus the final version of RTP proposal should not impact Core. Sorry if it caused any confusion.

Actually I'm not sure why the latest proposal doc includes Core as impacted platform. My last revision of the proposal removed it. (Please refer to this commit: https://github.com/smartdevicelink/sdl_evolution/commit/d5620e521ee2add9c511c6e48a5c6367e522b552)

theresalech, 2017-08-31:
We've confirmed that the RTP streaming changes are only needed on the Proxy and HMI side.

closed 2017-08-31 by theresalech