smartdevicelink/sdl_core #1564: SDL sends empty helpPrompt after ResetGlobalProperties

DrachenkoAnastasiia, 2017-05-26:
## Occurrence
Always

## Preconditions
1. SDL and HMI are running.
2. Register new application and activate it. 
3. Set GlobalProperties (helpPrompt, timeoutPrompt).

## Steps to reproduce
Reset GlobalProperties.

## Expected result
SDL sends on hmi timeoutPrompt and helpPrompt with default values.

## Actual result
SDL sends correct timeoutPrompt and empty helpPrompt.

_**Note:**_ 
Problem can be reproduced with [test_script](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/smoke_api.lua).

# Environment
Bug was reproduced on actual environment:
SDL repo: integration/4.3.1: https://github.com/LuxoftAKutsan/sdl_core/tree/integration/4.3.1
SDL commit 35344c8f9f249a1d6d5c4b4cfb2cd78d63fe3e4c
ATF repo: https://github.com/smartdevicelink/sdl_atf/tree/develop
ATF commit: fe002fb0e4e79cf9487f9c8bda6af7d61198a26d
HMI commit: N/A
OS: Ubuntu 14.04 LTS
Virtual machine: No
Transport: N/A
Mobile device: N/A
Mobile OS: N/A
Mobile App version: N/A
Mobile App type (media/non-media/navi): N/A

dboltovskyi, 2017-10-13:
According to req-t "[ResetGlobalProperties] SUCCESS on UI.SetGlobalProperties and TTS.SetGlobalPrtoperties":
>In case mobile app sends valid ResetGlobalProperties with "HELPPROMPT" and/or "TIMEOUTPROMPT" and at least one other valid parameter in "properties" array, SDL must:
>1) set-up the value of 'HelpPrompt' to an empty array and/or retrieve the value of "TimeOutPrompt" parameter from .ini file correspondingly
>2) transfer the items requested ("HELPPROMPT" or/and "TIMEOUTPROMPT") TTS.SetGlobalProperties(helpPrompt:"<empty array>", and/or timeoutPrompt:<'TimeOutPrompt' from ini.file>) to HMI.
>3) transfer UI.SetGlobalProperties with appropriate values to HMI (see more APPLINK-20718, APPLINK-20825).
>4) respond (resultCode:SUCCESS, success:true) to mobile application on getting the both SUCCESS:UI.SetGlobalProperties and SUCCESS:TTS.SetGlobalProperties from HMI

Value of "helpPrompt" has to be empty
So current issue is not actual

dboltovskyi, 2017-10-13:
Fix of automated script provided in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1702
Fix for SDL is closed: https://github.com/smartdevicelink/sdl_core/pull/1555

closed 2017-10-13 by dboltovskyi