smartdevicelink/sdl_core #1577: CreateInteractionChoiceSet: incorrect result code in case of duplicated "menuName"

DrachenkoAnastasiia, 2017-06-01:
## Occurrence
Always

## Preconditions
1. SDL is built with Proprietary flag.
2. SDL and HMI are started. First ignition cycle.
3. Connect device
4. Register new application.
5. Put icon.png file using PutFile RPC

## Steps to reproduce
1. Activate application.
2. Send CreateInteractionChoiceSet RPC with duplicated menuName within a choiceSet.

## Actual result
SDL->App: "resultCode: GENERIC_ERROR_NAME, success: false".

## Expected result
SDL->App: "resultCode: DUPLICATE_NAME, success: false".

# Environment
SDL: https://github.com/LuxoftAKutsan/sdl_core/tree/integration/4.3.1 (35344c8)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (fe002fb)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (4ba374c)
OS: Ubuntu 14.04 LTS x64

# Attachments
[SDLOPEN-2432.zip](https://github.com/smartdevicelink/sdl_core/files/1044641/SDLOPEN-2432.zip)

# Expected delivery   
- [ ] Source code updates 
- [ ] Code comments 
- [ ] UTs add/update 
- [ ] ATF tests add/update ***(not required)***  
- [x] Manual tests ***(not required)*** 
- [x] Add/update CI plans/jobs ***(not required)***  
- [x] SDD updates ***(not required)***  
- [x] Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) ***(not required)***  
- [x] Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines))  ***(not required)***

jacobkeeler, 2018-03-08:
Tested on master, `DUPLICATE_NAME` is received for any case where there is a duplicate name in `vrCommands`, but a `SUCCESS` response is received if two `menuName` fields are the same. It appears that there is no check for duplicates with `menuName`.

jacobkeeler, 2018-03-08:
Scenario I described is covered by #1032, closing this issue as duplicate.

closed 2018-03-08 by jacobkeeler