smartdevicelink/sdl_core #1578: SDL sends not valid RegisterAppInterface response with empty "imageTypeSupported"

dsmyrnov, 2017-06-01:
*Environment:*
OS: Ubuntu 16.04. 
ARC: PC (x86) 
SDL - Tag 4.2.1 commit e98cef965c4db1e3d9f1b225067571727ef1ecea
*Occurrence:*
Always
*Preconditions:*
SDL is running and HMI is connected to SDL
*Steps to reproduce:*
1) Register any application with SDL
2) Check RegisterAppInterface response
*Actual result:*
SDL respond with not valid empty "imageTypeSupported" 
*Expected result:*
SDL respond with valid RegisterAppInterface response
Requirements: MOBILE_API.xml

jacobkeeler, 2018-03-08:
Note, this seems to occur only if UI.GetCapabilities fails, leaving SDL Core to default to the capabilities in [hmi_capabilities.json](https://github.com/smartdevicelink/sdl_core/blob/master/src/appMain/hmi_capabilities.json). Not certain what the exact solution should be in this case, as this field cannot be omitted, due to it being mandatory. The only way to do so without changing the API would just be to omit every `ImageField` with this empty array.

iCollin, 2021-03-31:
Closed as no longer an issue now that the RAI response uses Window Capabilities which has `imageTypeSupported` with `minsize="0"`
Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin