smartdevicelink/sdl_core #1581: SubscribeVehicleData/UnsubscribeVehicleData: general result is "success":"true" instead of "false" in IGNORED notification for "DATA_ALREADY_SUBSCRIBED" parameters

DrachenkoAnastasiia, 2017-06-01:
## Occurrence
Always

## Preconditions
1. Mobile application is connected with SDL Core
2. SyncProxyTester is in BACKGROUND/FULL
3. Vehicle data parameter "prndl" is allowed by policies(but not subscribed), "VehicleInfo" group consented by user.

## Steps to reproduce
1. Send VehicleDataSubscriptions with "prndl".
2. Send same VehicleDataSubscriptions request with "prndl".

## Actual result
SubscribeVehicleData() response with resultCode:"IGNORED" and general result is success:"true".

## Expected result
SubscribeVehicleData() response with resultCode:"IGNORED" and general result success:"false" should come.

# Environment
SDL: https://github.com/LuxoftAKutsan/sdl_core/tree/integration/4.3.1 (35344c8)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (fe002fb)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (4ba374c)
OS: Ubuntu 14.04 LTS x64

# Attachments
[2439.zip](https://github.com/smartdevicelink/sdl_core/files/1044681/2439.zip)

# Expected delivery   
- [ ] Source code updates 
- [ ] Code comments 
- [ ] UTs add/update 
- [ ] ATF tests add/update ***(not required)***  
- [x] Manual tests ***(not required)*** 
- [x] Add/update CI plans/jobs ***(not required)***  
- [x] SDD updates ***(not required)***  
- [x] Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) ***(not required)***  
- [x] Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines))  ***(not required)***

JackLivio, 2018-05-30:
@DrachenkoAnastasiia Could you help me find some documentation or requirement that describes when the `success` parameter should be returned as `true` or `false`?

It is not clear to me which result codes correspond to success true or false. I am not certain that this issue's described behavior is incorrect.

GetmanetsIrina, 2018-05-31:
@JackLivio , This case is related to requirement
```
In case
app already subscribed on a single or multiple <vehicleData>
and sends SubscribeVehicleData_request with all <vehicleData> subscribed by the application before,
SDL must:
1) not transfer a resuest to HMI
2) respond with (IGNORED, success:false {<vehicleData>: DATA_ALREADY_SUBSCRIBED} ) to mobile app
```

The general requirement is 
```
In case SubscribeVehicleData is allowed by policies with less than supported by protocol parameters AND the app assigned with such policies requests SubscribeVehicleData with one and-or more allowed params and with one and-or more NOT-allowed params, SDL must:
1) transfer the allowed params of SubscribeVehicleData to HMI
2) get the response with <general_result_code_from_HMI> and allowed parameters with their corresponding individual-result-codes from HMI
3) respond to mobile application with "ResultCode: <general-result-code_from_HMI>, success: <applicable flag>" + "info" parameter(listing the params disallowed by policies and the information about allowed params processing) + allowed parameters and their corresponding individual result codes got from HMI and all disallowed parameters with the individual resultCode of DISALLOWED for NOT-allowed params of SubscribeVehicleData

Note: The response to SubscribeVehicleData with one of resultCodes of INVALID_DATA, OUT_OF_MEMORY, TOO_MANY_PENDING_REQUESTS, APPLICATION_NOT_REGISTERED should not contain any individual resutlCodes, so the requirement is not applicable to these resultCodes.

Note2: success:true and is applicable for <general_result_Code_from_HMI> SUCCESS or /WARNINGS only

Note3: In case on of the parameters is already subscribed by the application see requirements for IGNORED result code ( see above )
```

jacobkeeler, 2018-05-31:
@GetmanetsIrina Where is this requirement publicly available?

JackLivio, 2018-05-31:
@GetmanetsIrina 
Regarding this note:
```
Note2: success:true and is applicable for <general_result_Code_from_HMI> SUCCESS or /WARNINGS only
```

Is the expected behavior for all RPC's?

I am trying to find what the general rule for the success parameter is.

GetmanetsIrina, 2018-05-31:
@jacobkeeler , you can find requirements in https://github.com/smartdevicelink/sdl_requirements/blob/feature/FuelRange/detailed_docs/TRS/vehicle_info/SubscribeVehicleData_TRS.md

GetmanetsIrina, 2018-05-31:
@JackLivio about SUCCESS code yes.
Also about WARNING result code
```
In case SDL receives WARNINGS result code from HMI, SDL must transfer WARNINGS (success:true) to mobile app

Information:
WARNINGS result code is applicable for the following RPCs:
SystemRequest
AddCommand
Show
Speak
SubscribeVehicleData
UnsubscribeVehicleData
DeleteInteractionChoiceSet
CreateInteractionChoiceSet
AddSubMenu
DeleteSubMenu
SetAppIcon
Slider
ScrollableMessage
PerformAudioPathThru
Alert
AlertManeuver
PerformInteraction
DeleteCommand
SetGlobalProperties

HMI Important information:
HMI must respond with WARNINGS resultCode in case:
1. One or more requested items are not available (e.g. VehicleData for SubscribeVehicleData/UnsubscribeVehicleData )
2. Parameter(s) with ttsChunk type referenced is an unsupported phoneme
3. Parameter(s) with referenced type "imageType" is not supported on HMI
```
With `success:true` can be the next result codes:
SUCCESS
WARNINGS
WRONG_LANGUAGE
RETRY
SAVED
UNSUPPORTED_RESOURCE

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this issue. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this issue.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-05-19:
Hello, I am unable to reproduce this issue on Core 6.1.

I have created an ATF test to verify here: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2409

Please let me know if you are still seeing this issue and I will be happy to investigate.

Thank you

closed 2020-05-19 by iCollin