smartdevicelink/sdl_core #1585: AlertManeuver: SDL sends success = false in case of UNSUPPORTED_RESOURCE

DrachenkoAnastasiia, 2017-06-01:
## Occurrence
Always

## Preconditions
1. SDL is built with Proprietary flag.
2. SDL and HMI are started. First ignition cycle.
3. Connect device
4. Register new application.

## Steps to reproduce
1. App->SDL: AlertManeuver(imageType = "STATIC")
2. SDL->HMI: Navigation.AlertManeuver(imageType = "STATIC")
3. HMI->SDL: Response for Navigation.AlertManeuver(UNSUPPORTED_RESOURCE)
4. SDL-App: Response for AlertManeuver
5. Check parameters of AlertManeuver response 

## Actual result
{ success = false, resultCode = "UNSUPPORTED_RESOURCE"} 

## Expected result
{ success = true, resultCode = "UNSUPPORTED_RESOURCE"}

# Environment
SDL: https://github.com/LuxoftAKutsan/sdl_core/tree/integration/4.3.1 (35344c8)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (fe002fb)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (4ba374c)
OS: Ubuntu 14.04 LTS x64

# Attachments
[SDLOPEN-2451.zip](https://github.com/smartdevicelink/sdl_core/files/1044960/SDLOPEN-2451.zip)

# Expected delivery   
- [ ] Source code updates 
- [ ] Code comments 
- [ ] UTs add/update 
- [ ] ATF tests add/update ***(not required)***  
- [x] Manual tests ***(not required)*** 
- [x] Add/update CI plans/jobs ***(not required)***  
- [x] SDD updates ***(not required)***  
- [x] Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) ***(not required)***  
- [x] Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines))  ***(not required)***

jacobkeeler, 2018-03-08:
Related: #1596, #1597, #1600

jacobkeeler, 2020-09-29:
Not reproducible on develop

closed 2020-09-29 by jacobkeeler