smartdevicelink/sdl_core #1596: PerformInteraction: SDL sends success = false in case of UNSUPPORTED_RESOURCE

DrachenkoAnastasiia, 2017-06-07:
## Occurrence
Always

## Preconditions
1. SDL is built with Proprietary flag.
2. SDL and HMI are started. First ignition cycle.
3. Connect device
4. Register new application.
5. App->SDL: CreateInteractionChoiceSet() --> choice set is successfully created

## Steps to reproduce
1. App->SDL: PerformInteraction()
SDL->HMI:VR.PerformInteraction()
SDL->HMI:UI.PerformInteraction()
2. Responses
HMI->SDL: SUCCESS: UI.PerformInteraction()
HMI->SDL: UNSUPPORTED_RESOURCE: VR.PerformInteraction()
3. SDL->App: response

## Actual result
{ success = false, resultCode = "UNSUPPORTED_RESOURCE"} 

## Expected result
{ success = true, resultCode = "UNSUPPORTED_RESOURCE"}

# Environment
SDL: https://github.com/LuxoftAKutsan/sdl_core/tree/integration/4.3.1 (35344c8)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (fe002fb)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (4ba374c)
OS: Ubuntu 14.04 LTS x64

# Attachments
[SDLOPEN-2453.zip](https://github.com/smartdevicelink/sdl_core/files/1057903/SDLOPEN-2453.zip)

# Expected delivery   
- [ ] Source code updates 
- [ ] Code comments 
- [ ] UTs add/update 
- [ ] ATF tests add/update ***(not required)***  
- [x] Manual tests ***(not required)*** 
- [x] Add/update CI plans/jobs ***(not required)***  
- [x] SDD updates ***(not required)***  
- [x] Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) ***(not required)***  
- [x] Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines))  ***(not required)***

jacobkeeler, 2020-09-29:
Not reproducible on develop

jacobkeeler, 2020-12-21:
Closing with the merge of #3587

PR #3587 (2020-12-21): Fix success=false after UNSUPPORTED_RESOURCE response for multiple RPCs
Fixes #1596, #1597, #1583

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Run ATF Tests in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2494

### Summary
Fix issues with response parameter calculation after UNSUPPORTED_RESOURCE response for PerformInteraction, Alert, SubtleAlert, AlertManeuver, and PerformAudioPassThru

### Changelog
##### Bug Fixes
* Fix success=false after UNSUPPORTED_RESOURCE response for several RPCs
* Fix success, choiceID, and triggerSou [... cut]
closed 2020-12-21 by jacobkeeler